arduino/arduino-pro-ide #278: Function println not matched when using ArduinoCore-samd core

enekochan, 2020-06-20:
**Describe the bug**
Using the [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd) `println` function in `Serial` is highligthed as "No matching member function for call to '`println`'" even if it is shown in code completition and the sketch is built without problems:

```
No matching member function for call to 'println'

/Users/username/Library/Arduino15/packages/arduino/hardware/samd/1.8.6/cores/arduino/Print.h:78:12:
note: candidate function not viable: no known conversion from 'Uart' to 'Print' for object argument
```

**To Reproduce**
1. Install ArduinoCore-samd.
2. Create a simple sketch and write a `Serial.println("...");` statement
3. See error

**Expected behavior**
Call to `println` be correctly recognized as belonging to `Serial`.

**Screenshots**
[image]

**Desktop:**
 - OS: macOS 10.15.5
 - Version: Arduino Pro 0.0.6 (0.0.6.20200409.2)
 - ArduinoCore-samd 1.8.6

ubidefeo, 2021-02-18:
@enekochan 
This should be resolved, can you try the 0.1.4?
[image]

enekochan, 2021-02-18:
Yes it works fine :)

enekochan, 2021-02-18:
Sorry no, still not working.

[image]

enekochan, 2021-02-18:
Full error log:

```
No matching member function for call to 'println'

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:79:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:78:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:77:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:80:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:81:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:82:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:83:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:84:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:85:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:86:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:87:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:88:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:89:12:
note: candidate function not viable: no known conversion from 'Uart' to 'arduino::Print' for object argument

/Users/enekochan/Library/Arduino15/packages/arduino/hardware/samd/1.8.11/cores/arduino/api/Print.h:90:12:
note: candidate function not viable: requires 0 arguments, but 1 was provided
```

ubidefeo, 2021-02-18:
thanks @enekochan 
we'll look for a solution :)