arduino/arduino-pro-ide #279: Error: arduino-beta:mbed:envie_m4 is not a valid FQBN.

aseitasi15, 2020-06-22:
**Describe the bug**
try to build for Arduino Portenta H7, it says "Error: arduino-beta:mbed:envie_m4 is not a valid FQBN." 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "new sketch"
2. Click on 'build'

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac Mojave
 - Version 10.14.6

per1234, 2020-06-22:
Hi @aseitasi15. The alpha stage Arduino Pro IDE doesn't currently support FQBNs (fully qualified board name) that contain `-` (https://github.com/arduino/arduino-pro-ide/issues/272). Because the Portenta uses the same architecture name, `mbed` as the Nano 33 BLE, Arduino is currently using a temporary vendor name `arduino-beta` for the Portenta, so that the initial releases of the Portenta's board platform are optional for Nano 33 BLE users who have that board's dedicated platform installed already (Nano 33 BLE is also supported by the board platform used by the Portenta, so both boards will eventually use the same board platform). The side effect of this is that the Portenta's temporary FQBN contains `-`, making it not work with the Arduino Pro IDE. The Arduino developers are aware of this and working to add support to the Pro IDE for hyphenated FQBNs.

The easiest way to get started using your Portenta is with the classic Arduino IDE, which you can download here:
https://www.arduino.cc/en/Main/Software

If you prefer to use Arduino Pro IDE, I can provide you with instructions for a workaround that will allow you to use the Portenta with it.

aseitasi15, 2020-06-22:
Yes please show me the workaround. I really interested in the Arduino PRO IDE in this time. thx.

per1234, 2020-06-23:
OK, here you go:

1. Rename `/Users/{username}/Library/Arduino15/packages/arduino-beta` to  `/Users/{username}/Library/Arduino15/packages/arduinobeta`
1. (In the Arduino Pro IDE) **File > Settings > Open CLI Configuration**
1. To the `additional_urls` field of the `arduino-cli.yaml` file that is opened, add the URL: `https://gist.githubusercontent.com/per1234/e49b8ff532b1617d880a9f08298cada9/raw/b57d7c97f788344a569afedc730439ec81375a0f/package_portentaworkaround_index.json` as shown below:
    ```yaml
    board_manager:
      additional_urls: [https://gist.githubusercontent.com/per1234/e49b8ff532b1617d880a9f08298cada9/raw/b57d7c97f788344a569afedc730439ec81375a0f/package_portentaworkaround_index.json]
    ```
1. Wait for the "Index has been updated." message to be shown at the bottom right corner of the Pro IDE window.
1. Close all Arduino Pro IDE windows.
1. Start the Pro IDE again.

---
**In case you're interested, an explanation of the above instructions:**
The ["vendor folder"](https://arduino.github.io/arduino-cli/latest/platform-specification/#hardware-folders-structure)'s name determines the first component of the FQBN, so the rename changes the Portenta's FQBN from `arduino-beta:mbed:envie_m7` to `arduinobeta:mbed:envie_m7` (which will allow you to use the Portenta with the Arduino Pro IDE without getting the "is not a valid FQBN" error).

In order for the Pro IDE to recognize the Arduino mbed-enabled boards platform in the new path, it is necessary to provide it with a modified package index file. I published that package index at the long URL in the instructions. The index is simply a copy/paste of the Arduino mbed-enabled boards section of [Arduino's official package index](https://downloads.arduino.cc/packages/package_index.json), with the vendor name field changed from `arduino-beta` to `arduinobeta`.

aseitasi15, 2020-06-23:
Thank you so much. it works finally.

closed 2020-06-23 by aseitasi15