arduino/arduino-pro-ide #281: Add libraries that not included in Library manager

TrungNgo2809, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
Sometimes I want to use a library that not included in Library manager but can't find the source libraries folder to add that library.

**Describe the solution you'd like**
In normal Arduino folder, there is another folder called "libraries " and if I want to add some libraries that not included in Library manger, I'll just have to add a file that include it's source code to the "libraries " folder and that solve the proplem.
So I hope the developers can add that funtion to Arduino Pro IDE.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Still love Arduino Pro for it's Convinience and Beautifull layout, much appreciate the developers for that.

Erriez, 2020-09-10:
I've created lost of Arduino libraries which are all compatible with Arduino Classic IDE and PlatformIO, but I can't find a menu to install 3rd party boards and libraries via the "pro" IDE.

Can you please implement menus to a GitHub URL or ZIP file to extract them automatically into the `Arduino/libraries` directory:
1. Board Manager
2. Library Manager

ubidefeo, 2020-09-10:
@Erriez 
our LIbrary managers go through a central library repository.
Your way to install a custom library is to add it in the `libraries` subfolder of your sketch folder.
That works for everyone.

You can already install a custom board by adding the index to your `additional_urls` entries, just point them to the `.json` file defining your boards according to the specs.

Erriez, 2020-09-10:
@ubidefeo Thanks for your reply.

> our LIbrary managers go through a central library repository.
I'd like to add some of my custom libraries to this repository as well. I'll check how to do this. (Different topic)

I found the custom board installation via `File | Open CLI Configuration`:  
```
board_manager:
  additional_urls: []
```

> Your way to install a custom library is to add it in the libraries subfolder of your sketch folder.
My idea is to support `.json` URL's to custom libraries as well. My libraries already contain a `.json` file for PlatformIO, maybe this can be reused. 

Is this possible?

per1234, 2020-09-10:
>I'd like to add some of my custom libraries to this repository as well. I'll check how to do this.

Thanks! You'll find all the information here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

>My idea is to support .json URL's to custom libraries as well. My libraries already contain a .json file for PlatformIO, maybe this can be reused.

Just to link things together to make it easy to find all the discussion on the topic, here is a similar feature request in the Arduino IDE's repository:
https://github.com/arduino/Arduino/issues/6879
Note that request is for allowing the URLs to custom library_index.json files, not URLs directly to individual library metadata files, as is done by PlatformIO.

Erriez, 2020-09-10:
Thanks for the links, @per1234. That's very useful.

ubidefeo, 2020-09-10:
thank you for chiming in, @per1234 
I was about to reply but had to jump into a webinar 🙏

per1234, 2021-02-17:
TrungNgo2809's issue was resolved.

There is a dedicated issue at https://github.com/arduino/arduino-pro-ide/issues/128 for the lack of parity with the classic Arduino IDE's **Sketch > Include Library > Add ZIP Library...** (as reported by Erriez).

So I'll go ahead and close this.

closed 2021-02-17 by per1234