arduino/arduino-pro-ide #282: Serial monitor speed limiter to 115200 (legacy ID

njz3, 2020-06-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kittaakos, 2020-06-30:
> Compared to legacy IDE's serial monitor, it would be nice to get the "extra" baudrate like 230400, 250000, 500000, 1000000, 2000000,

I propose the following:
 - Introduce a new preference for additional, user-defined baud rates,
 - By default, there aren't any custom baud rates,
 - Users can add custom ones, this custom baud rate shows up in the UI with the built-in ones,
 - Users can select it.

Pros:
 - We do not have to update the code when we have to add support for a new baud rate, users can do it locally,
 - We do not pollute the UI with all the baud rates.

Cons:
 - The implementation requires more effort than just adding a new enum.

Thoughts?

cmaglie, 2020-06-30:
Custom baudrate are not always supported and from the Java IDE we know that very few user needs it (well actually no one ever asked for a custom baudrate as far as I remember).

IMHO it's a lot of work for very little benefit, for now I'll just add the missing "standard" baudrate and see how it goes, @kittaakos we may implement your proposal in the future if there is demand.

njz3, 2020-06-30:
I may be one of the few user that use the "extra" baudrates ;-)
Those are useful for example for the Mega2560, the Leonardo or the Due where the COM port would handle much more than the 115200 baudrate. Specifically, the Mega2560 does not handle well 115200 (timing errors), while 250000 and 500000 are perfectly well handled.

A mixed solution would be to add also a 'custom' baudrate in the list which would use a configurable value in the preference, or simply use an editable combobox to enter a custom value instead of a fixed list of choices.

kittaakos, 2020-06-30:
> * a new preference for additional

Just to make sure we're all on the same page: I meant a Pro IDE preference. It has nothing to do with the CLI config.

cmaglie, 2020-06-30:
> I may be one of the few user that use the "extra" baudrates ;-)

In fact I said to add them.

> A mixed solution would be to add also a 'custom' baudrate in the list which would use a configurable value in the preference, or simply use an editable combobox to enter a custom value instead of a fixed list of choices.

Did you use a baudrate that is not already listed in the Java IDE? what's the use case?

cmaglie, 2020-06-30:
(I'm asking because we have a huge backlog for the pro IDE and don't want to spend time on features that are very minor improvement)

njz3, 2020-06-30:
I personnally use the standard baudrates of the Java IDE, not a custom baudrate.
To help you sorting out, I would say:
- must: please add at least the same baudrates as the Java IDE.
- minor/nice to have: please add a "custom" entry where the value would be taken from the preference. Some classical baudrates like 115200 are not valid for an Atmega@16Mhz (see the baudrate calculator here http://ruemohr.org/~ircjunk/avr/baudcalc/avrbaudcalc-1.0.8.php?postbitrate=9600&postclock=16&bit_rate_table=on).

njz3, 2020-06-30:
I forgot to explain the use case: 
I use an Atmega2560 where I implemented a basic text protocol to control my system (I do motor control). The terminal can return lots of information after a command is interpreted. Most of the time, I directly type the commands from the Arduino IDE to test my code and send commands. I can switch my terminal to a streaming mode with a message every "X" ms with lots of debug printout and I can use the serial plotter to draw internal variables. It is very handy.
(The very same protocol is also used in a communication library to perform realtime control of a motor.)

Using the 115200 baudrate with the Mega2560 was giving garbage characters to appear from time to time or wrong point on the serial plotter, furthermore it was limiting how many values I can output every milliseconds. 
This was really annoying, that's why I moved to 1000000 baudrate now.