arduino/arduino-pro-ide #283: How to get started

zfred68, 2020-06-26:
I downloaded the package for linux, there's a shared library file arduino-pro-ide. Then what? There is no readme file
listing what pre-requisites are required, or INSTALL.txt of how to install. So I just learned something about Theia, sure
so it seems I need to catchup with some stuff, but how far down the rabbit hole must one go to figure out how to try this
out. Is it too much to ask for whoever to describe the steps to get going. Oh i've read the github head page, a few times,
its still not helping.

**Is your feature request related to a problem? Please describe.**
A It is a problem when there are no clear instructions for some new tech that is happening, very frustrating.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Attigliuzzo, 2020-06-26:
On my debian testing is quite simple: unzip in a folder and use it. Be aware when you unzip, there could be missing library, they are still there. Use a good program to extract it (file roller in my case), i had issues extracting with windows or xarchiver, it missed symlink of several files.

zfred68, 2020-06-26:
ok! It showing up as a shared library file type threw me off.feel stupid now. should there be a readme or install file with the package though?

hpssjellis, 2020-07-19:
@zfred68 don't feel stupid. I have tons of experience and just did the exact same thing. Came here looking for why nothing worked. Will try a better unzip program, but nothing working on my ubuntu 20.04 and no readme except the original suggestions to look for an "install" file that isn't there. A very short README.txt would have saved me a bit of frustration.

Suppasit, 2020-08-21:
I would like to share my frustration here. 
I am trying to run Arduino Pro IDE v0.0.6 on my Ubuntu 18.04 LTS. With no instruction I tried to run it as a regular user on my machine. The problem is that the IDE gave me the error of unable to find sketch file inside my /tmp/ folder, and constantly refused to compile or doing anything (the code suggesting also doesn't work).
But when running the application with sudo command, the IDE seems to work OK. Hope there will be more clear instructions on how to run and install this IDE on Linux machine soon.