arduino/arduino-pro-ide #284: installation Arduino IDE Pro

nivatec, 2020-06-26:
has version for Windows7 32 bits ??
i'm trying to install but i can't.

kittaakos, 2020-06-29:
> has version for Windows7 32 bits ??
> i'm trying to install but i can't.

No, unfortunately, we do not (yet) support 32bit Windows.

I am not closing this issue, but the 32bit arch support is out of the context of the `alpha` release cycles.
See: https://github.com/arduino/arduino-pro-ide/issues/138#issuecomment-568408919

tuxedo0801, 2020-10-01:
@nivatec Why don't you update to a 64bit os? What kind of old hardware do you have which requires 32bit os?

nivatec, 2020-10-06:
boa tarde amigo Alex
o meu PC é fraco e não posso comprar outro  e esse que tenho não aceita o sistema 64bit
obrigado pela sua atenção.
-Brazil-

good afternoon friend Alex
my PC is weak and I cannot buy another one and the one I have does not accept the 64bit system
Thank you for your attention.