arduino/arduino-pro-ide #286: I wish there was documentation about how to change font size

AndySantaCruzAnalytics, 2020-07-04:
very hard to figure out how to change preferences. A little documentation would go a long way.

The pro version is much better than the standard IDE

for example, I want to make the font bigger

kind regards

Andy

Attigliuzzo, 2020-07-05:
If you mean the font size in the input code workspace, you just need to edit "editor->fontSize" variable, type a new value.

AndySantaCruzAnalytics, 2020-07-05:
many thanks

The pro IDE is much better. I realize it is still under development and so documentation is kind of sparse. Do you know of a maybe a quick your tube video that talks about configuration options. I find the current mechanism overwhelming. 

One of the things I can not do is close files? My project has a lot of files. I only want to keep a small number of them open. This reduces accidental typos

Andy

Attigliuzzo, 2020-07-05:
I agree with you, i can successfully use this instead of the classic one. You don't need videos or guides, simply use it; you will spend time to understand it but it is better that classic,

Master811129, 2020-07-05:
you can also use Ctrl + MouseWeel to change the font

kittaakos, 2021-02-11:
We have re-implemented the Java IDE-based settings. You can increase the font size there. It's also possible from the `Edit` menu or from the keyboard `Ctrl/Cmd++` and `Ctrl/Cmd+-`. Closing.

closed 2021-02-11 by kittaakos