arduino/arduino-pro-ide #292: All colors are faded, hard to see, how can I adjust them myself?

selevo, 2020-07-25:
**Describe the bug**
All colors are faded, hard to see, how can I adjust them myself?

OS WIN 7

[image]

kittaakos, 2020-07-27:
You can pick another color theme:
 - Open the _Command Palette_ (<kbd>F1</kbd> or <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>)
 - Execute the `Preferences: Color Theme` command,
 - Pick another theme.

I hope this helps.

[image]

closed 2020-07-27 by kittaakos

Master811129, 2020-07-27:
> You can pick another color theme:
> 
> * Open the _Command Palette_ (F1 or Ctrl/⌘+Shift+P)
> * Execute the `Preferences: Color Theme` command,
> * Pick another theme.
> 
> I hope this helps.
> 
> [image]

WOW so many theme! can we see them in the next release?

selevo, 2020-07-29:
Thanks, but I asked how to change the color myself, I was not interested in other people's schemes.

And second, I only have 4 color schemes, what am I doing wrong?
[image]

kittaakos, 2020-07-29:
> Thanks, but I asked how to change the color myself, I was not interested in other people's schemes.

This is not yet ~supported~ available, but I do not see any technical obstacles; the idea is the following:
 - The Arduino Pro IDE is based on [Eclipse Theia](https://theia-ide.org/),
 - Theia can run any VS Code extensions,
 - In theory, you can install and run any VS Code extensions, including themes,
 - You can [craft](https://code.visualstudio.com/api/extension-guides/color-theme#create-a-new-color-theme) your own color theme and publish it to for instance [Open VSX registry](https://open-vsx.org/),
 - You can install and consume it from the Pro IDE.

> And second, I only have 4 color schemes, what am I doing wrong?

Nothing. The latest publicly available release was rolled out months ago.