arduino/arduino-pro-ide #293: Path too long

sjh114961, 2020-07-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kittaakos, 2020-07-28:
> Getting a path too long unzipping.

There is not much we can do about it, try to extract the app to `C:\`. Let us know if it still does not work.

sjh114961, 2020-07-28:
I installed it on the "c:\" and that work. The answer is to shorten your name to a much more reasonable size. Now I get 
this when I try to debug.
[image]

alranel, 2020-07-29:
@sjh114961 what path where you initially trying to unzip in?
(Please open a distinct issue for the debugger thing)

alranel, 2020-07-29:
I see that the file with the longest name in the Windows package is `./resources/app/plugins/vscode-yaml/extension/node_modules/yaml-language-server/out/server/node_modules/vscode-languageserver-protocol/node_modules/vscode-languageserver-types/tsconfig.esm.json` which is 194 characters long. @kittaakos do you think that we can't really do much about this?

By the way, maybe would this be solved when we stop distributing the application as a zip file and we provide a proper  installer instead?

kittaakos, 2020-07-29:
> do you think that we can't really do much about this?

Unfortunately, yes. This path comes from a VS Code extension for the YAML language as-is: https://github.com/redhat-developer/vscode-yaml

We cannot change the paths, as Node.js might [`require`](https://nodejs.org/en/knowledge/getting-started/what-is-require/) them when it runs the VS Code extensions. I will ask around in the Theia community just to make sure I am not missing something and will update this thread.

If the problem occurs only when unzipping the app and not at runtime:
 - a proper app installer could help,
 - or as a temporary workaround, here are a few threads describing how to enable long path support on Windows 10:
   - https://thegeekpage.com/fix-error-0x80010135-path-too-long
   - https://superuser.com/questions/1119883/windows-10-enable-ntfs-long-paths-policy-option-missing
   - https://winaero.com/blog/how-to-enable-ntfs-long-paths-in-windows-10

kittaakos, 2020-07-31:
As for now, we decided to drop the YAML language support from the application. This means the users will lose [all the language features](https://github.com/arduino/arduino-pro-ide/issues/245) (validation, content assist, hover) in the CLI config editor. This is unfortunate, but we have to replace the CLI config editor with some proper UI anyways.

I have opened an upstream issue: https://github.com/arduino/arduino-pro-ide/issues/245

CC: @ubidefeo @rsora