arduino/arduino-pro-ide #295: 1.2.2 Dropped - Portanta Not Recognized

jwestmoreland, 2020-07-28:
Hello,

Is there an update to this:

[https://gist.githubusercontent.com/per1234/e49b8ff532b1617d880a9f08298cada9/raw/b57d7c97f788344a569afedc730439ec81375a0f/package_portentaworkaround_index.json

The 1.2.2 download installed - 1.8.13 ok but Pro-IDE doesn't recognize the Portenta using the above that worked with 1.2.1.

Thanks,
John

per1234, 2020-07-28:
Hi @jwestmoreland. Unfortunately, the workaround breaks on every release of the Arduino mbed-Enabled Boards platform.

Here's the new Boards Manager URL:
https://gist.githubusercontent.com/per1234/e49b8ff532b1617d880a9f08298cada9/raw/af4158932e1537bc90bbc5c741340765eeb2c2c4/package_portentaworkaround_index.json

Just replace the previous one in your File > Settings > Open CLI Configuration, wait until you see the popup notification "The CLI configuration has been successfully reloaded", then restart the Pro IDE.

After that, the Portenta should again be recognized. I'll get the other instructions about this workaround updated with the new URL.

If you still have a problem, let me know and I'll help you out.

ubidefeo, 2020-07-28:
@jwestmoreland 
bear with us, we're close to releasing an update to the alpha IDE that will handle this annoyance once and for all.
it's a matter of days now.
As development moves forward we'll have a more defined release schedule :)

jwestmoreland, 2020-07-28:
@per1234 and @ubidefeo ,

pert & Ubi - Thank you both for the quick response!  Looking forward to the new alpha IDE release too!

Best Regards,
John W.

jwestmoreland, 2020-07-28:
@per1234 ,

I'm getting an error - please see the below:
Error:  Request getBoardDetails failed with message:  13 INTERNAL:
Failed to parse server response

[image]

Regards,
John W.

per1234, 2020-08-02:
Hi @jwestmoreland sorry for the slow response. I don't know what could be the cause of this
```
Request getBoardDetails failed with message: 13 INTERNAL: Failed to parse server response
```
error. I don't have any problems when using my Portenta with the Pro IDE and the workaround package index URL.

Do you have any other Arduino boards? If so, do you get the same error when you use them with the Pro IDE?

jwestmoreland, 2020-08-02:
pert,

Just tried with the Nano IoT and it does the same thing.

I used to work with a bunch of guys that would say "works for you and not for me" - I guess that's the case.

Due to what was added to #295 I thought I had to wait for 0.7.0 - guess I'm glad I asked now.

So, suggestions?  I'm definitely using this:
`[https://gist.githubusercontent.com/per1234/e49b8ff532b1617d880a9f08298cada9/raw/af4158932e1537bc90bbc5c741340765eeb2c2c4/package_portentaworkaround_index.json`

Regards,
John

per1234, 2020-08-02:
What happens if you run this Arduino CLI command with any of the boards connected to your computer:
```
arduino-cli board list
```

I am suspicious this error happens when Arduino CLI attempts to contact the builder.arduino.cc API to try to auto-detect the board based on its VID/PID pair:
https://github.com/arduino/arduino-cli/blob/0.11.0/commands/board/list.go#L38

jwestmoreland, 2020-08-02:
C:\tmp>arduino-cli board list
Port  Type              Board Name                    FQBN                       Core
COM11 Serial Port (USB) Unknown
COM12 Serial Port (USB) Unknown
COM16 Serial Port (USB) Unknown
COM17 Serial Port (USB) Unknown
COM22 Serial Port (USB) Unknown
COM23 Serial Port (USB) Arduino NANO 33 IoT           arduino:samd:nano_33_iot   arduino:samd
COM4  Serial Port (USB) Arduino Portenta H7 (M7 core) arduino-beta:mbed:envie_m7 arduino-beta:mbed
COM5  Serial Port (USB) Unknown
COM6  Serial Port (USB) Unknown
COM7  Serial Port (USB) Unknown

jwestmoreland, 2020-08-03:
Pert,

Update:

I also checked my routers and firewalls just to see if something was going on there - but nothing showed up regarding this issue.

But, I did find this:

M$ Defender had been blocking network access of arduino-cli.exe - it allowed access for the one in my path which was not the same as the one for the Pro IDE.

I have since fixed that - I thought - but I will probably either try to disable defender completely or uninstall it now to see if I can get the problem resolved.

I had no idea that defender was even blocking access for this app at all.  I have other virus scan/firewall tools that I thought were in sole control - but seems M$ feature creep is omnipresent.

Regards,
John

jwestmoreland, 2020-08-04:
Pert,

Well, this just keeps getting interesting.

I thought I had disabled M$ Defender - note, with the new Win10 2020 Updates - the normal 'google fu' on how to do that doesn't apply - things are a little 'new' after some of the updates - I got the following:

[image]

Regards,
John

jwestmoreland, 2020-08-20:
> @jwestmoreland
> bear with us, we're close to releasing an update to the alpha IDE that will handle this annoyance once and for all.
> it's a matter of days now.
> As development moves forward we'll have a more defined release schedule :)

Hmmm, matter of days....

ubidefeo, 2020-08-21:
@jwestmoreland 
I know, I know.
We bumped into a few show-stoppers and to be honest a release like that would have upset a lot of users.
We prefer to get a snarky (deserved/appreciated) comment like yours than release something that won't make us happy.
Suffice it to say we have already written the release notes ;)

jwestmoreland, 2020-08-21:
@ubidefeo ,

Is the next release going to be 0.0.7?   And, has there been any progress on the debugger?  (ThreadLib?)

Thanks and looking forward to it!

Best Regards,
John W.

ubidefeo, 2020-08-26:
@jwestmoreland 
> I don't like tools that compile C to be held hostage by .json. It's a productivity killer - plain and simple.

fair.

> And for all of the java-only people out there - a report earlier this year put 'C' back at the top of computer languages.
I come from ECMA (not considering C64 Basic), plenty of years developing in ActionScript for interactive products, but then discovered C/C++ and it changed my mindset.
I have done a decent amount of Java and J2ME: pure hate.

good luck exploring :)

jwestmoreland, 2020-08-26:
Hello @ubidefeo ,

> fair.

More than.

Also,

> I come from ECMA (not considering C64 Basic), plenty of years developing in ActionScript for interactive products, but then discovered C/C++ and it changed my mindset.
> I have done a decent amount of Java and J2ME: pure hate.
> 
> good luck exploring :)

assuming all of those are your comments.

Exploring going ok from the debugger side:
[image]
[image]

Regards,
John

jwestmoreland, 2020-08-28:
Hello @ubidefeo ,

Another interesting thing worth showing - when setup() gets called - eclipse will start the Arduino IDE - note, AFAIK, I haven't installed or enabled any Arduino plug-ins in eclipse even though in earlier versions I had that installed but see now they are deprecated - I'm running the 2020-09 M2 version of Eclipse.  I checked under installed software and didn't see anything that said 'Arduino'.

[image]

Regards,
John W.

ubidefeo, 2020-08-28:
unless you have some piece of the puzzle calling the app this cannot be but a poltergeist.
how have you set this up to run?

jwestmoreland, 2020-08-28:
Hello @ubidefeo ,

> unless you have some piece of the puzzle calling the app this cannot be but a poltergeist.

Ha!  Wow - that's funny.  I did have some of the previous plug-ins installed for Arduino - I guess there's a chance eclipse is pulling something in - not sure - but hopefully it isn't a poltergeist!

> how have you set this up to run?

It's allegedly just plain-Jane C/C++ IDE set up.  I've set up openocd and gdb fairly 'classically'.  The installation details are here:

[image]

Regards,
John

ubidefeo, 2020-08-29:
I think I know which _previous plugin_ you're talking about, and that one calls the Java IDE to compile everything, hence the app being launched.
It definitely is a launch target thing, which Arduino Eclipse plugin do you have installed?

jwestmoreland, 2020-08-29:
Hello @ubidefeo ,

Can you see the image in my last post - that's what's installed.  Maybe you can't see that - I've posted images in other forums on github where people couldn't see the images I posted in the replies.

In a previous installation - I see the C/C++ Arduino Tools are installed - I guess that's what is being accessed.

Regards,
John W.

ubidefeo, 2020-08-29:
@jwestmoreland 
I see the list of installed things, but can't spot any Arduino plugin for Eclipse in that list.
This makes me wonder what can possibly launch the IDE during a run 🤔

jwestmoreland, 2020-08-29:
Yes @ubidefeo , I know.  The only explanation I have is that a previous installation had the Arduino plugins, and even though there are in separate directories - it must be accessing it even though it doesn't show being installed.

This is previous install that I'm still using (for ESP32 stuff mostly.)

[image]

Thanks,
John W.

ubidefeo, 2020-08-30:
There, at the top, lies that ancient grave you built your Eclipse env upon.
That's the cause of your poltergeist :D

jwestmoreland, 2020-08-30:
Hello @ubidefeo ,

Indeed - especially since that's not the version I run with the poltergeist appears.  ;)

Thanks,
John W.

closed 2020-12-21 by per1234