arduino/arduino-pro-ide #296: Examples

AndyTo-tech, 2020-07-30:
Hello and congratulations for this Arduino Pro IDE.!
I am a user of the old application and something i noticed is missing is the menu "File->Examples". I could not find it even in the "classic" (nor togglin into the Advanced mode)
For newbies like me this was really helpfull from the beginning.
I hope you can add it. Thank a lot!
Andy

ubidefeo, 2020-07-31:
hi @AndyTo-tech 
Examples are coming, they're very important not only for new users.
I have been using Arduino since 2006 and I still open that menu and load examples to start from :D

A bit of patience, they'll be there by the time this is publicly available and not just an `alpha` ✌️

closed 2020-07-31 by ubidefeo

AndyTo-tech, 2020-07-31:
Thanks a lot!! :)