arduino/arduino-pro-ide #298: macOS: Command + r is hijacked globally

j-rieck, 2020-08-07:
**Describe the bug**
When Arduino Pro IDE is running in the background, <kbd>⌘</kbd> <kbd>R</kbd> is hijacked globally, preventing other applications from utilising it.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Arduino Pro IDE
2. Hit <kbd>⌘</kbd> <kbd>R</kbd> on the keyboard
3. Watch as nothing happens
4. Close Arduino Pro IDE
5. Repeat step 2.
6. The page will now reload

**Expected behavior**
It is expected that <kbd>⌘</kbd> <kbd>R</kbd> will only trigger within Arduino Pro IDE when the application is the currently active one.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.15.5
 - Arduino Pro IDE 0.0.6 (0.0.6.20200409.2)

**Additional context**
Captured with ShortcutDetective

j-rieck, 2020-08-07:
I just checked out the eclipe-theia repository, and it seems there was a change three months ago in [this  commit](https://github.com/eclipse-theia/theia/commit/8ca3e6459362320f26f80a7e072eacc7d09ceb4a).

"Removed global shortcuts if window is not focused." Is the Arduino Pro IDE running on a version before this commit? If so, that would explain the issue.

ubidefeo, 2020-08-07:
hi @j-rieck 

first of al: great icon!
on to the serious stuff: we have fixed this in the meantime, and a new realease is on its way with many more changes.
We're planning a better schedule for updates on this project, but decided to first address most of the annoying things before pushing a new one out.

The wait is not gonna be long, we're days away :)

j-rieck, 2020-08-07:
Moof!

Thanks for the update, looking forward to the release, and the eventual release of the source code.

Thanks for your hard work!

kittaakos, 2020-08-10:
@j-rieck, [here](https://github.com/arduino/arduino-pro-ide/issues/249#issuecomment-628531565) is a workaround if you cannot wait until the next release.

j-rieck, 2020-08-10:
Thanks @kittaakos. I didn't think of searching closed issued before opening this one.

kittaakos, 2021-02-11:
Fixed. Closing.

closed 2021-02-11 by kittaakos