arduino/arduino-pro-ide #299: Incompatibility with Arduino CLI 0.11.0

LumitoLuma, 2020-08-15:
**Describe the bug**
The program does not work properly if you have Arduino CLI 0.11.0 installed on your computer.

**To Reproduce**
Steps to reproduce the behavior:
1. Download Arduino CLI 0.11.0
2. Add it to PATH variable
3. Open Arduino Pro IDE
4. You will receive some alerts while you use the program and you will not be able to compile/upload the project.

**Expected behavior**
I excepted that the program would work perfectly with Arduino CLI 0.11.0 installed.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2004

**Additional context**
I do not have this issue with Arduino CLI 0.10.0.

ubidefeo, 2020-08-16:
hi @LumitoLuma 

For the time being the Pro IDE does not make use of a standalone installed Arduino CLI, but bundles its own.
This is because during the alpha phase we might continue to refine API and gRPC interfaces, so we pin a specific version of the CLI to be bundled with the Pro IDE.

We're very close to a new release of Pro IDE which will have CLI version higher than 0.10.
I assume you've figured out a workaround to use an other CLI running in daemon mode by changing the port, but this is not currently supported/endorsed because of said changes in gRPC interfaces which might just break things.

Please have a few more days of patience and follow this space for a release of both CLI and IDE :)

LumitoLuma, 2020-08-24:
Issue closed since release [0.1.0](https://github.com/arduino/arduino-pro-ide/releases/0.1.0) fixes this error. Thanks!

closed 2020-08-24 by LumitoLuma