arduino/arduino-pro-ide #300: Cannot upload a program to a card that does not require port selection.

philibertc, 2020-08-17:
Hello, 

I own the MH-Tiny (MH-ET LIVE) card which requires the connection to the computer of this one after the compilation of the program only. It is then the programmer (Micronucleus) that takes care of finding the right USB device.
Here is my problem:
The Arduino Pro Ide prevents me from compiling and then uploading my code without having selected a serial port.
Does anyone have the solution?
Thanks.

ubidefeo, 2020-08-17:
hi @philibertc 
it's a bug, we are aware of this, and it's fixed in the upcoming release (trust me, it's coming very soon).

good news is that you can select any random serial port and the recipe won't use it, but the programmer will still be selected if the core and platform have correct scripts :)

give it a go
u.

philibertc, 2020-08-17:
Hello,
Thank you very much for your response.
I have one last question: how can we choose a specific programmer and microcontroller?

ubidefeo, 2020-08-17:
@philibertc 
You cannot directly choose a microcontroller and a programmer unless they are defined as specs of the core platform.
If you download a core definition set (for instance our own AVR) when selecting a target board the programmer menu will populate according to the available recipes.

You can create your own custom boards and cores with all the recipes you want or search the web to see what's available.

We do not provide custom implementations of 3rd party cores, but 3rd parties develop boards and cores definitions for all kinds of microcontrollers with all kinds of setups.

Here's a great starting point https://arduino.github.io/arduino-cli/latest/platform-specification/

Hope it helps get you started
u.

philibertc, 2020-08-18:
Okay, thanks, I'll look this.

closed 2020-08-18 by philibertc