arduino/arduino-pro-ide #301: Does the arduino pro ide debugger supports arduino DUE?

wanggaoteng, 2020-08-18:
Hope it supports arduino DUE.

ubidefeo, 2020-08-18:
@wanggaoteng 
debugging functionalities are enabled based on platform definition, and the Due does not have those fields filled in.
you could try and modify the `platform.txt` to support it and is if it fits your needs.

This is the block related to debug for `samd` platforms, it might get you started.
(Please remember that debug support in Arduino Pro IDE is still very experimental)

```
#
# GDB (Debugger)
#
# EXPERIMENTAL feature: debug.pattern
#  - this is alpha and may be subject to change without notice
tools.gdb-openocd.path={runtime.tools.arm-none-eabi-gcc-7-2017q4.path}/bin/
tools.gdb-openocd.cmd=arm-none-eabi-gdb
tools.gdb-openocd.cmd.windows=arm-none-eabi-gdb.exe
tools.gdb-openocd.interpreter=console
tools.gdb-openocd.debug.pattern="{path}/{cmd}" --interpreter={interpreter} -ex "set remotetimeout 5" -ex "set pagination off" -ex 'target extended-remote | "{tools.openocd.path}/{tools.openocd.cmd}" -s "{tools.openocd.path}/share/openocd/scripts/" --file "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "gdb_port pipe" -c "telnet_port 0"' "{build.path}/{build.project_name}.elf"
```

wanggaoteng, 2020-08-24:
> @wanggaoteng
> debugging functionalities are enabled based on platform definition, and the Due does not have those fields filled in.
> you could try and modify the platform.txt to support it and is if [...]

Thank you @ubidefeo 
By the way, when will the stable official verson of arduino-pro-ide be released?
Platformio is a greate project, it is very suitable for arduino, does Arduino have a chance to work with Platformio? For creating a more powerful IDE.

ubidefeo, 2020-08-26:
>By the way, when will the stable official verson of arduino-pro-ide be released?
we just released `0.1.0` yesterday and it's pretty stable for an Alpha.
We're moving fast but don't have an ETA for `1.0.0` yet :) 

> Platformio is a greate project, it is very suitable for arduino, does Arduino have a chance to work with Platformio? For creating a more powerful IDE.
PlatformIO and Arduino are two distinct projects which I don't see converging at any time in the future.
Our tools are slowly being refactored for a larger audience and I hope to see someone switch back from PlatformIO to Arduino when our product offering is more aligned to what such users need.

In the meantime we can only work towards having a product we like to use internally and our users can contribute to with suggestions, bug reports and... a little bit of love for the philosophy that made Arduino the first and largest Open Source hardware and firmware project.
We like to think that we have inspired hundreds of talented people into creating outstanding projects :)

ubidefeo, 2021-02-18:
This issue has been transferred into the SAM core repo and will hence be closed
https://github.com/arduino/ArduinoCore-sam/issues/118

closed 2021-02-18 by ubidefeo