arduino/arduino-pro-ide #304: Unable to Launch IDE

mattb1969, 2020-08-24:
Fantastic that somebody is willing to develop a decade IDE for Arduino, one that contains features like debugger. I wish I was that talented to be able to do it myself.

I’ve just downloaded V0.0.6 from the repository for Mac OSX and when I try to run the application my security settings don't allow it to run as the developer can’t be verified.

Steps to reproduce the behavior:
1. Download IDE for Mac OSX by clicking on link in /releases
2. Once downloaded, unpack into applications folder
3. Start Application
4. See error

I can change my security settings, but it is not possible to verify the package so it will be accepted by Apple?

kittaakos, 2020-08-24:
Thanks for logging, @mattb1969. Can you please try with the [`0.1.0` version](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.0)? The macOS app has been signed and notarized.

mattb1969, 2020-08-25:
Issue resolved, software installed and opened fine

mattb1969, 2020-08-25:
resolved - mac accepted software

closed 2020-08-25 by mattb1969