arduino/arduino-pro-ide #306: v 0.1.0 problems with 3rd party cores (MegaCoreX)

sstaub, 2020-08-24:
**Describe the bug**
- I can't upload to Curiosity Nano 4809 board using MegaCoreX. Fails with ERROR 2 (see screenshot). The core works fine the standard Arduino IDE 1.8.13.
Error from AVRDude: avrdude: Can't find programmer id "{upload.protocol}"
- In the screenshot you see a warning for the board, however.
- I get some unexpected error with Intellisense (#include Arduino.h, Serial3.print())
- When starting the IDE I get many errors (see the attached messages)

**To Reproduce**
Install MegaCoreX
Select ATmega4809
Select Programmer Atmel nEDGB
Upload simple blinkt example

**Expected behavior**
Upload should work with the programmers given by the 3rd party core

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Catalina
 - Version 0.1.0

**Additional context**

**I get many errors when starting the IDE:**

[INFO] Checking signature [index: /Users/sstaub/Library/Arduino15/package_index.json, signaturefile: /Users/sstaub/Library/Arduino15/package_index.json.sig, trusted: true]
[INFO] Checking signature [error: opening signature file: open /Users/sstaub/Library/Arduino15/package_stm_index.json.sig: no such file or directory, index: /Users/sstaub/Library/Arduino15/package_stm_index.json, signaturefile: /Users/sstaub/Library/Arduino15/package_stm_index.json.sig]
[INFO] Checking signature [error: opening signature file: open /Users/sstaub/Library/Arduino15/package_MCUdude_MegaCoreX_index.json.sig: no such file or directory, index: /Users/sstaub/Library/Arduino15/package_MCUdude_MegaCoreX_index.json, signaturefile: /Users/sstaub/Library/Arduino15/package_MCUdude_MegaCoreX_index.json.sig]
[INFO] Checking if CLI is Bundled into the IDE
[INFO] Loading hardware from: /Users/sstaub/Library/Arduino15/packages
[INFO] Loading package ATTinyCore from: /Users/sstaub/Library/Arduino15/packages/ATTinyCore/hardware
[INFO] Loaded platform [platform: {ID:"ATTinyCore:avr",Installed:"1.4.1",Latest:"1.4.1"}]
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/ATTinyCore/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/ATTinyCore/tools
[INFO] Loaded tool [tool: Version: 2.0a4]
[INFO] Loading package DxCore from: /Users/sstaub/Library/Arduino15/packages/DxCore/hardware
[INFO] Loaded platform [platform: {ID:"DxCore:megaavr",Installed:"1.1.0",Latest:"1.1.0"}]
[INFO] Loading package MegaCoreX from: /Users/sstaub/Library/Arduino15/packages/MegaCoreX/hardware
[INFO] Loaded platform [platform: {ID:"MegaCoreX:megaavr",Installed:"1.0.4",Latest:"1.0.4",Name:"MegaCoreX"}]
[INFO] Loading package STM32 from: /Users/sstaub/Library/Arduino15/packages/STM32/hardware
[INFO] Loaded platform [platform: {ID:"STM32:stm32",Installed:"1.9.0",Latest:"1.9.0",Name:"STM32 Cores"}]
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/STM32/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/STM32/tools
[INFO] Loaded tool [tool: Version: 5.5.1 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: all]]
[INFO] Loaded tool [tool: Version: 1.4.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loaded tool [tool: Version: 9.2.1-1.1 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loading package arduino from: /Users/sstaub/Library/Arduino15/packages/arduino/hardware
[INFO] Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.3",Latest:"1.8.3",Name:"Arduino AVR Boards"}]
[INFO] Loaded platform [platform: {ID:"arduino:megaavr",Installed:"1.8.6",Latest:"1.8.6",Name:"Arduino megaAVR Boards"}]
[INFO] Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.8",Latest:"1.8.8",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/arduino/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/arduino/tools
[INFO] Loaded tool [tool: Version: 4.5.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Loaded tool [tool: Version: 1.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Loaded tool [tool: Version: 1.2.1 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 1.3.0 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loaded tool [tool: Version: 7.3.0-atmel3.6.1-arduino5 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 7.3.0-atmel3.6.1-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 6.3.0-arduino17 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin12, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 1.7.0-arduino3 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 0.10.0-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Loading package builtin from: /Users/sstaub/Library/Arduino15/packages/builtin
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/builtin/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/builtin/tools
[INFO] Loaded tool [tool: Version: 5.8-arduino11]
[INFO] Loaded tool [tool: Version: 1.0.0]
[INFO] Loading package chipKIT from: /Users/sstaub/Library/Arduino15/packages/chipKIT/hardware
[... 186 lines cut ...]
[INFO] Loaded tool [tool: Version: 5.8-arduino11]
[INFO] Loaded tool [tool: Version: 1.0.0]
[INFO] Loading package chipKIT from: /Users/sstaub/Library/Arduino15/packages/chipKIT/hardware
[INFO] Loaded platform [platform: {ID:"chipKIT:pic32",Installed:"2.1.0",Latest:"2.1.0"}]
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/chipKIT/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/chipKIT/tools
[INFO] Loaded tool [tool: Version: v2]
[INFO] Loaded tool [tool: Version: 1.43-pic32gcc]
[INFO] Loaded tool [tool: Version: v2.1.46]
[INFO] Loading package esp8266 from: /Users/sstaub/Library/Arduino15/packages/esp8266/hardware
[INFO] Checking existence of 'tools' path: /Users/sstaub/Library/Arduino15/packages/esp8266/tools
[INFO] Loading tools from dir: /Users/sstaub/Library/Arduino15/packages/esp8266/tools
[INFO] Loading package megaTinyCore from: /Users/sstaub/Library/Arduino15/packages/megaTinyCore/hardware
[INFO] Loaded platform [platform: {ID:"megaTinyCore:megaavr",Installed:"2.0.5",Latest:"2.0.5"}]
[INFO] Loading hardware from: /Users/sstaub/Documents/Arduino/hardware
[INFO] Adding libraries dir [dir: /Users/sstaub/Documents/Arduino/libraries, location: user]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/megaTinyCore/hardware/megaavr/2.0.5/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/DxCore/hardware/megaavr/1.1.0/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/chipKIT/hardware/pic32/2.1.0/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/STM32/hardware/stm32/1.9.0/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.4.1/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/arduino/hardware/samd/1.8.8/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/sstaub/Library/Arduino15/packages/MegaCoreX/hardware/megaavr/1.0.4/libraries, location: platform]

sstaub, 2020-08-24:
Ok, I found that I must use "Upload Using Programmer"
However the errors when starting persist I will open a new issue

kittaakos, 2020-08-24:
Looks like the index update has failed:
```
rror while updating index in attempt 8: Error: 2 UNKNOWN: invalid package index in http://drazzy.com/package_drazzy.com_index.json: invalid patch version separator 'a'[INFO] Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
Updating index: package_index.json completed.
Updating index: package_index.json.sig completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167.sig]
[INFO] Updating index [url: https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json]
Updating index: package_stm_index.json completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465.sig]
[INFO] Updating index [url: https://mcudude.github.io/MegaCoreX/package_MCUdude_MegaCoreX_index.json]
Updating index: package_MCUdude_MegaCoreX_index.json completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524.sig]
[INFO] Updating index [url: http://drazzy.com/package_drazzy.com_index.json]
Updating index: package_drazzy.com_index.json completed.
```

Does it work if you install the 3rd party core with the CLI?

closed 2020-08-24 by kittaakos