arduino/arduino-pro-ide #308: DxCore is not shown in the board manager

sstaub, 2020-08-24:
**Describe the bug**
The board manager does not show the cores from SpenceKonde
[https://github.com/SpenceKonde]
There are the DxCore and some Tiny cores.
The board manager works in the standard Arduino IDE 1.8.13

**To Reproduce**
Steps to reproduce the behavior:
1. add `http://drazzy.com/package_drazzy.com_index.json` to CLI config
2. restart the IDE I get many error messages when loading, see issue #306 
3. The cores are not seen in the board manager

**Expected behavior**
The core should also work with the Pro IDE

**Desktop (please complete the following information):**
- OS: macOS Catalina
- Version 0.1.0

kittaakos, 2020-08-24:
From [here](https://github.com/arduino/arduino-pro-ide/issues/306#issuecomment-679305561):
```
rror while updating index in attempt 8: Error: 2 UNKNOWN: invalid package index in http://drazzy.com/package_drazzy.com_index.json: invalid patch version separator 'a'[INFO] Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
Updating index: package_index.json completed.
Updating index: package_index.json.sig completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/055000167.sig]
[INFO] Updating index [url: https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json]
Updating index: package_stm_index.json completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/293532465.sig]
[INFO] Updating index [url: https://mcudude.github.io/MegaCoreX/package_MCUdude_MegaCoreX_index.json]
Updating index: package_MCUdude_MegaCoreX_index.json completed.
[INFO] Checking signature [error: opening signature file: open /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524.sig: no such file or directory, index: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524, signaturefile: /var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/703650524.sig]
[INFO] Updating index [url: http://drazzy.com/package_drazzy.com_index.json]
Updating index: package_drazzy.com_index.json completed.
```

per1234, 2020-08-24:
Thanks so much for bringing this to everyone's attention @sstaub!

Since the code that causes this is in Arduino CLI, I'm going to close it in favor of the report of the same issue in the arduino/arduino-cli repository: https://github.com/arduino/arduino-cli/issues/920. Feel free to subscribe to that issue to follow the discussion and provide any feedback you might have.

closed 2020-08-24 by per1234