arduino/arduino-pro-ide #309: Upload button don't works when using a programmer

sstaub, 2020-08-24:
**Is your feature request related to a problem? Please describe.**
The are now two Uplaods commands
- Upload
- Upload Using Programmer
It is confusing because there is no symbol in the Pro IDE for "Upload Using Programmer"
In the standard IDE there is only one button for both, but also two commands.
See issue #306 where the problem comes up using MegaCoreX

**Describe the solution you'd like**
The upload button / command should work also for "Upload Using Programmer"

ubidefeo, 2020-08-24:
@sstaub 
Thank you for being so prompt with filing this issue.
It slipped through the cracks, like other ones I'm sure you'll find :)

per1234, 2020-08-25:
The problem is the Arduino Pro IDE doesn't support this behavior:
https://arduino.github.io/arduino-cli/dev/platform-specification/#upload-using-programmer-by-default
>If the `upload.protocol` property is not defined for a board, the Arduino IDE's "Upload" process will use the same behavior as "Upload Using Programmer". This is convenient for boards which only support uploading via programmer.

If you look at sstaub's screenshot from #306, you can see the FQBN of the selected board has the `bootloader` custom board option set to `no_bootloader`. For that configuration, the platform author intentionally did not define an `upload.protocol` property because a standard "Upload" would not be useful for a board with no bootloader, and thus it is convenient to give the "Upload" button the "Upload Using Programmer" behavior, which is how the Arduino IDE works:
https://github.com/MCUdude/MegaCoreX/blob/v1.0.4/megaavr/boards.txt#L151-L167