arduino/arduino-pro-ide #310: 0.1 default color scheme sucks (near useless as everything is too dark)

paschlie, 2020-08-24:
Open ide, and discover it's nearly illegible.

kittaakos, 2020-08-24:
@paschlie, can you please attach a screenshot to the issue?

kittaakos, 2020-08-24:
By default, the IDE comes with a (light) Arduino theme. Although we made a few UI changes, the theme should be the same. You can try to change it: 
 - Open the _Command Palette_ with <kbd>Cltr/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>,
 - Type: `Color Theme`,
 - Press enter,
 - Select another theme.

paschlie, 2020-08-24:
> Open the Command Palette with Cltr/⌘+Shift+P,

The 0.1 IDE as installed this morning defaulted to Dark, I changed the appearance preference to “light”, and quit and restarted the app, now it just hangs with a blue colored circle circling around the center of the displayed window with nothing else appearing; so ended up just quitting and trying again after abut 15 minutes with the same results.

(It it helps, I’m running MACOS 10.13.6 on a 10 year old MacBook Pro, which doesn’t inherently support a “Dark” theme; and tried re-installing with the same results; so am continuing using the basic IDE 1.8.14 nightly build for the time being.)

ubidefeo, 2020-08-26:
@paschlie 
the IDE does not rely on the OS dark mode support but implements colour schemes.
The process described by @kittaakos (command palette and theme) should just work in the app and even preview the theme.
I haven't experienced the default to dark behaviour on Mojave or Catalina, but don't have a High Sierra machine to test on.
I'll try and setup a VM for testing and see if I can reproduce

paschlie, 2020-08-26:
Please find following two screen shots of this morning’s nightly builds attempting to start up, but apparently stuck doing something; with the second showing the apparent default to “dark” mode with the preference panel being displayed after shift/cmd/P being entered ... (so am at a loss on how to progress further?)

ubidefeo, 2020-08-27:
@kittaakos 
can you look into this?
any chance you have a High Sierra lying around?

kittaakos, 2020-08-27:
From the attached screenshots, I do not see how this issue is related to the theme, but the app should load.

@paschlie, what do you see in the console when you start the app from a terminal:
```
/Applications/Arduino\ Pro\ IDE.app/Contents/MacOS/Arduino\ Pro\ IDE 
```

My output starts as:
```
root WARN please install @theia/electron@1.5.0-next.09ef844b as a runtime dependency
root INFO Configuration directory URI: 'file:///Users/akos.kitta/.theia'
root INFO Configuring to accept webviews on '.+.webview..+' hostname.
root INFO Theia app listening on http://localhost:64533.
config INFO Started watching the Arduino CLI configuration: '/Users/akos.kitta/.arduinoProIDE/arduino-cli.yaml'.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.44.2" with engines: { vscode: '*' }
root INFO Deploying backend plugin "cpp@1.44.2" from "/Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploy plugins list took: 34.8 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 64.3 ms
daemon INFO Starting daemon from /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Using config file: /Users/akos.kitta/.arduinoProIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.12.1
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:10000
daemon INFO Daemon is now listening on 127.0.0.1:10000...
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:64538.
daemon INFO Checking signature [index: /Users/akos.kitta/Library/Arduino15/package_index.json, signaturefile: /Users/akos.kitta/Library/Arduino15/package_index.json.sig, trusted: true]
daemon INFO Checking signature [error: opening signature file: open /Users/akos.kitta/Library/Arduino15/package_esp32_index.json.sig: no such file or directory, index: /Users/akos.kitta/Library/Arduino15/package_esp32_index.json, signaturefile: /Users/akos.kitta/Library/Arduino15/package_esp32_index.json.sig]
daemon INFO Checking signature [error: opening signature file: open /Users/akos.kitta/Library/Arduino15/package_adafruit_index.json.sig: no such file or directory, index: /Users/akos.kitta/Library/Arduino15/package_adafruit_index.json, signaturefile: /Users/akos.kitta/Library/Arduino15/package_adafruit_index.json.sig]
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /Users/akos.kitta/Library/Arduino15/packages
daemon INFO Loading package adafruit from: /Users/akos.kitta/Library/Arduino15/packages/adafruit/hardware
daemon INFO Loaded platform [platform: {ID:"adafruit:avr",Installed:"1.4.12",Latest:"1.4.13",Name:"Adafruit AVR Boards"}]
daemon INFO Loading package arduino from: /Users/akos.kitta/Library/Arduino15/packages/arduino/hardware
daemon INFO Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.3",Latest:"1.8.3",Name:"Arduino AVR Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:mbed",Installed:"1.1.6",Latest:"1.1.6",Name:"Arduino nRF528x Boards (Mbed OS)"}]
daemon INFO Loaded platform [platform: {ID:"arduino:megaavr",Installed:"1.8.6",Latest:"1.8.6",Name:"Arduino megaAVR Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.8",Latest:"1.8.8",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
daemon INFO Checking existence of 'tools' path: /Users/akos.kitta/Library/Arduino15/packages/arduino/tools
daemon INFO Loading tools from dir: /Users/akos.kitta/Library/Arduino15/packages/arduino/tools
```

paschlie, 2020-08-27:
Mine was:

root WARN please install @theia/electron@1.5.0-next.09ef844b as a runtime dependency
root INFO Configuration directory URI: 'file:///Users/paul/.theia'
root WARN Backend Object.initialize is slow, took: 54.4 ms
root WARN Backend ElectronTokenBackendContribution.configure is slow, took: 51.5 ms
root INFO Configuring to accept webviews on '.+.webview..+' hostname.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/plugins does not exist.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/extensions does not exist.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/plugins does not exist.
root INFO Theia app listening on http://localhost:55517.
config INFO Started watching the Arduino CLI configuration: '/Users/paul/.arduinoProIDE/arduino-cli.yaml'.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.44.2" with engines: { vscode: '*' }
config ERROR Error occurred when loading CLI config from /Users/paul/.arduinoProIDE/arduino-cli.yaml. YAMLException: can not read a block mapping entry; a multiline key may not be an implicit key at line 4, column 77:
     ... ndex/package_adafruit_index.json
                                         ^
    at generateError (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:167:10)
    at throwError (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:173:9)
    at readBlockMapping (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1073:9)
    at composeNode (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1359:12)
    at readBlockMapping (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1089:11)
    at composeNode (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1359:12)
    at readDocument (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1525:3)
    at loadDocuments (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1588:5)
    at load (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1614:19)
    at Object.safeLoad (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/js-yaml/lib/js-yaml/loader.js:1637:10)
root WARN Backend ConfigServiceImpl.onStart is slow, took: 95.8 ms
daemon INFO Starting daemon from /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
root INFO Deploying backend plugin "cpp@1.44.2" from "/Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploy plugins list took: 512.5 ms
daemon INFO Error reading config file: While parsing config: yaml: line 4: could not find expected ':'
daemon INFO Failed to start the daemon.
daemon ERROR Error: Error reading config file: While parsing config: yaml: line 4: could not find expected ':'
    at Socket.<anonymous> (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:322:58)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
daemon INFO Restarting daemon in 5 seconds...
daemon INFO time="2020-08-27T07:32:35-04:00" level=error msg="Error reading config file: While parsing config: yaml: line 4: could not find expected ':'"
daemon INFO Using config file: /Users/paul/.arduinoProIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.12.1
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Restarting daemon in 4 seconds...
daemon INFO Restarting daemon in 3 seconds...
daemon INFO Restarting daemon in 2 seconds...
daemon INFO Restarting daemon in 1 seconds...
daemon INFO Restarting daemon now...
daemon INFO Starting daemon from /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Error reading config file: While parsing config: yaml: line 4: could not find expected ':'
daemon INFO Failed to start the daemon.
daemon ERROR Error: Error reading config file: While parsing config: yaml: line 4: could not find expected ':'
    at Socket.<anonymous> (/Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:322:58)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
daemon INFO Restarting daemon in 5 seconds...
[... cut ...]
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon ERROR listen tcp :9090: bind: address already in use
daemon ERROR Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Daemon exited with exit code: 5.
daemon INFO Restarting daemon in 4 seconds...
daemon INFO Restarting daemon in 3 seconds...
daemon INFO Restarting daemon in 2 seconds...
daemon INFO Restarting daemon in 1 seconds...
daemon INFO Restarting daemon now...
^C

kittaakos, 2020-08-27:
> config ERROR Error occurred when loading CLI config from /Users/paul/.arduinoProIDE/arduino-cli.yaml. YAMLException: can not read a block mapping entry; a multiline key may not be an implicit key at [...]

This is very helpful. The app could not start, because the CLI configuration is invalid. Of course, this is a bug and the IDE must be smarter to handle invalid configuration files gracefully. The config is at ` ~/.arduinoProIDE/arduino-cli.yaml`.

Can you please share its content with use. Mine looks like this:
```bash
cat ~/.arduinoProIDE/arduino-cli.yaml 
```

```yaml
board_manager:
  additional_urls: [
    'https://dl.espressif.com/dl/package_esp32_index.json',
    'https://adafruit.github.io/arduino-board-index/package_adafruit_index.json'
  ]
daemon:
  port: "10000"
directories:
  data: /Users/akos.kitta/Library/Arduino15
  downloads: /Users/akos.kitta/Library/Arduino15/staging
  user: /Users/akos.kitta/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
```

There are two ways to solve this issue:
 - you can try to correct it manually,
 - or delete the file and the IDE will recreate it with the defaults.

I can  see another problem from the logs:
```
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon ERROR listen tcp :9090: bind: address already in use
```
There is a telemetry port that is used by some other application, so the CLI just crashes at runtime. Can  you  check what's running on port 9090 on your mac?

Thank you for your persistence in figuring out the root cause of the defect. 👍

paschlie, 2020-08-27:
Mine was:

board_manager:
  additional_urls:
  https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
  https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

daemon:
  port: "50051"
directories:
  data: /Users/paul/Library/Arduino15
  downloads: /Users/paul/Library/Arduino15/staging
  user: /Users/paul/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true

----

I deleted the parent directory, and restarted 0.1.1, and it now runs again (thank you), it defaulted to the “Dark” theme (but changed it to “Arduino Light”), and restarted via terminal with the following output (the port issue seems to have taken care of itself?):

Last login: Thu Aug 27 07:46:55 on ttys000
----------------------------------------------------------------------- 08:00:27
paul@Fusion:~$ /Applications/Arduino\ Pro\ IDE.app/Contents/MacOS/Arduino\ Pro\ IDE 
root WARN please install @theia/electron@1.5.0-next.09ef844b as a runtime dependency
root INFO Configuration directory URI: 'file:///Users/paul/.theia'
root INFO Configuring to accept webviews on '.+.webview..+' hostname.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/plugins does not exist.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/extensions does not exist.
root WARN The directory referenced by local-dir:/Users/paul/.arduinoProIDE/plugins does not exist.
root INFO Theia app listening on http://localhost:56082.
config INFO Started watching the Arduino CLI configuration: '/Users/paul/.arduinoProIDE/arduino-cli.yaml'.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.44.2" with engines: { vscode: '*' }
root INFO Deploying backend plugin "cpp@1.44.2" from "/Applications/Arduino Pro IDE.app/Contents/Resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploy plugins list took: 66.6 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 137.3 ms
daemon INFO Starting daemon from /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Using config file: /Users/paul/.arduinoProIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.12.1
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:56087.
daemon INFO Checking signature [index: /Users/paul/Library/Arduino15/package_index.json, signaturefile: /Users/paul/Library/Arduino15/package_index.json.sig, trusted: true]
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /Users/paul/Library/Arduino15/packages
daemon INFO Loading package Seeeduino from: /Users/paul/Library/Arduino15/packages/Seeeduino/hardware
daemon INFO Loaded platform [platform: {ID:"Seeeduino:samd",Installed:"1.7.9",Latest:"1.7.9"}]
daemon INFO Checking existence of 'tools' path: /Users/paul/Library/Arduino15/packages/Seeeduino/tools
daemon INFO Loading tools from dir: /Users/paul/Library/Arduino15/packages/Seeeduino/tools
daemon INFO Loaded tool [tool: Version: 5.4.0]
daemon INFO Loaded tool [tool: Version: 1.2.1]
daemon INFO Loaded tool [tool: Version: 1.2.1]
daemon INFO Loaded tool [tool: Version: 7-2017q4]
daemon INFO Loaded tool [tool: Version: 1.7.0-arduino3]
daemon INFO Loaded tool [tool: Version: 1.8.0-48-gb176eee]
daemon INFO Loading package adafruit from: /Users/paul/Library/Arduino15/packages/adafruit/hardware
daemon INFO Loaded platform [platform: {ID:"adafruit:samd",Installed:"1.6.0",Latest:"1.6.0"}]
daemon INFO Checking existence of 'tools' path: /Users/paul/Library/Arduino15/packages/adafruit/tools
daemon INFO Loading tools from dir: /Users/paul/Library/Arduino15/packages/adafruit/tools
daemon INFO Loaded tool [tool: Version: 5.4.0]
daemon INFO Loaded tool [tool: Version: 9-2019q4]
daemon INFO Loading package arduino from: /Users/paul/Library/Arduino15/packages/arduino/hardware
daemon INFO Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.8",Latest:"1.8.8",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
[... 174 lines cut ...]
daemon INFO Loading package espressif from: /Users/paul/Documents/Arduino/hardware/espressif
daemon INFO Loaded platform [platform: {ID:"espressif:esp32"}]
daemon INFO Adding libraries dir [dir: /Users/paul/Documents/Arduino/libraries, location: user]
daemon INFO Adding libraries dir [dir: /Users/paul/Library/Arduino15/packages/adafruit/hardware/samd/1.6.0/libraries, location: platform]
daemon INFO Adding libraries dir [dir: /Users/paul/Library/Arduino15/packages/Seeeduino/hardware/samd/1.7.9/libraries, location: platform]
daemon INFO Adding libraries dir [dir: /Users/paul/Documents/Arduino/hardware/espressif/esp32/libraries, location: platform]
daemon INFO Adding libraries dir [dir: /Users/paul/Library/Arduino15/packages/arduino/hardware/samd/1.8.8/libraries, location: platform]
root INFO [hosted-plugin: 12967] PLUGIN_HOST(12967): PluginManagerExtImpl/init()
root INFO [hosted-plugin: 12967] PLUGIN_HOST(12967): initializing(cpp@1.44.2 with /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/@theia/plugin-ext-vscode/lib/node/plugin-vscode-init.j [... cut ...]
discovery INFO Updated discovered boards and available ports. - Discovered boards: [] and available ports: [{"protocol":"serial","address":"/dev/cu.PaulsRugby-BluetoothSer"},{"protocol":"serial","address":"/dev/cu.SGH-A847-SerialServer"},{"protocol":"serial","address":"/dev/cu.Bluetooth-Incoming-Port"}]
root INFO Attached boards and available ports changed:  {"oldState":{"boards":[],"ports":[]},"newState":{"boards":[],"ports":[{"protocol":"serial","address":"/dev/cu.PaulsRugby-BluetoothSer"},{"protocol":"serial","address":"/dev/cu.SGH-A847-SerialServer"},{"protocol":"serial","address":"/dev/cu.Bluetooth-Incoming-Port"}]}}

----

kittaakos, 2020-08-27:
> Mine was:

Thanks! That's incorrect YAML:
```yaml
board_manager:
  additional_urls:
   - https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
   - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

daemon:
  port: "50051"
directories:
  data: /Users/paul/Library/Arduino15
  downloads: /Users/paul/Library/Arduino15/staging
  user: /Users/paul/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
```

or 
```yaml
board_manager:
  additional_urls: [
   'https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json',
   'https://adafruit.github.io/arduino-board-index/package_adafruit_index.json'
   ]

daemon:
  port: "50051"
directories:
  data: /Users/paul/Library/Arduino15
  downloads: /Users/paul/Library/Arduino15/staging
  user: /Users/paul/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
```

are the correct format.

paschlie, 2020-08-27:
> Mine was:
>
> Thanks! That's incorrect YAML:
>
> board_manager:
>   additional_urls:
>    - https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
>    - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
>
> daemon:
>   port: "50051"
> directories:
>   data: /Users/paul/Library/Arduino15
>   downloads: /Users/paul/Library/Arduino15/staging
>   user: /Users/paul/Documents/Arduino
> logging:
>   file: ""
>   format: text
>   level: info
> telemetry:
>   addr: :9090
>   enabled: true
> or
>
> board_manager:
>   additional_urls: [
>    'https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json',
>    'https://adafruit.github.io/arduino-board-index/package_adafruit_index.json'
>    ]
>
> daemon:
>   port: "50051"
> directories:
>   data: /Users/paul/Library/Arduino15
>   downloads: /Users/paul/Library/Arduino15/staging
>   user: /Users/paul/Documents/Arduino
> logging:
>   file: ""
>   format: text
>   level: info
> telemetry:
>   addr: :9090
>   enabled: true
> are the correct format.

How should additional board managers be added within the IDE?

kittaakos, 2020-08-27:
> How should additional board managers be added within the IDE?

Check this thread: https://github.com/arduino/arduino-pro-ide/issues/245#issue-613309368

It describes how to add additional, 3rd party URLs.

paschlie, 2020-08-27:
> Check this thread: #245 (comment) <https://github.com/arduino/arduino-pro-ide/issues/245#issue-613309368>

Thank you, please consider the issue closed (with the exception of anything you may have discovered during the dialog which should be further refined).

kittaakos, 2020-08-27:
> with the exception of anything you may have discovered during the dialog which should be further refined

Sure, I will create follow-ups regarding the CLI config issue.

I am so happy we managed to figure out the what was causig the problem. Thank you for giving us some time to find the solution. 

(I am going to close the task once the follow-up GH issue exists.)

paschlie, 2020-08-27:
As a closing thought, it might be nicer if third-party board support urls could be added within the ide’s board selection pane, as opposed to having to edit the configuration file (which if for some reason it may be added incorrectly, the application may prevented from starting up?).

kittaakos, 2020-08-27:
> As a closing thought, it might be nicer if third-party board support urls could be added within the ide’s board selection pane

Sure, I already created an feature request in the upstream framework: https://github.com/eclipse-theia/theia/issues/8081

Also, please note, we used to have proper validation in the YAML editor: https://user-images.githubusercontent.com/1405703/75560969-7a0b6880-5a46-11ea-935c-2adba1b28d8a.gif

But we had to remove because of this issue on Windows: https://github.com/arduino/arduino-pro-ide/issues/293#issuecomment-667139709

kittaakos, 2020-08-27:
Follow-up: https://github.com/arduino/arduino-pro-ide/issues/322

closed 2020-08-27 by kittaakos