arduino/arduino-pro-ide #311: Verify and Upload button doesn't work

Master811129, 2020-08-24:
**Describe the bug**
Verify and Upload button doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Arduino Pro IDE'
2. Click on 'Verify'
3. See the problem

**Desktop (please complete the following information):**
 - OS: [Windows]

Master811129, 2020-08-24:
Stange! when i moved Arduino Pro IDE folder into C drive it works properly

Master811129, 2020-08-24:
OK i tested 0.0.6. it works properly in another drive .that means this bug is new.

kittaakos, 2020-08-24:
Thank you, @Master811129. 👍 

Yes, that behavior has changed. We create the new sketches into the temp folder, then move them to the desired location on (first) save.

 - Can you please tell us what was the original path before you moved it to C?
 - What is the output of `echo %TEMP%` in a `CMD.exe`?

Master811129, 2020-08-25:
it was in D:/ before
here is %temp% in D:/
[image]
and in c:/
[image]

kittaakos, 2020-08-25:
I managed to reproduce the issue on my Windows machine and fix it. We try to include the fix within the next [nightly build](https://github.com/arduino/arduino-pro-ide#nightly-builds).

kittaakos, 2020-08-25:
I did not want to close the issue, merging the PR has automatically closed it so I am going to reopen. @Master811129, can you please close it if it works with the nighly? Thank you!

Master811129, 2020-08-27:
It works with official boards:
[image]
And also in 3rd party:
[image]
sorry for the delay

closed 2020-08-27 by Master811129