arduino/arduino-pro-ide #312: v0.1.0 - Using TheadDebug Lib:  GDB Debugger Terminated Unexpectedly.

jwestmoreland, 2020-08-24:
**Describe the bug**
Attempting to debug with the ThreadLib debug library - get message GDB debugger terminated unexpectedly.

**To Reproduce**
Steps to reproduce the behavior:
1. At the top of your code:
#include <ThreadDebug.h>

// UartDebugCommInterface debugComm(SERIAL1_TX, SERIAL1_RX, 230400);
// ThreadDebug threadDebug(&debugComm, DEBUG_BREAK_IN_SETUP);
UsbDebugCommInterface debugComm(&SerialUSB);
ThreadDebug threadDebug(&debugComm, DEBUG_NO_BREAK_IN_SETUP);

2. Build & Download
3. Start debug session by Debug->Start Debugging
4. See error - GDB debugger terminated unexpectedly.

**Expected behavior**
Debug session should start.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Message pops up and goes away.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
Windows 10 - 2020 latest released from M$

jwestmoreland, 2020-08-24:
From the command line - using what was compiled by the 0.1.0 IDE - it does run:

`C:\tmp\demo-web-server\build\arduinobeta.mbed.envie_m7>C:\Users\John\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\bin\arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote COM4" demo-web-server.ino.elf
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from demo-web-server.ino.elf...done.
Remote debugging using COM4
0x08061682 in osKernelResume (sleep_ticks=14) at ./mbed-os/rtos/source/TARGET_CORTEX/rtx5/RTX/Source/rtx_kernel.c:492
492     ./mbed-os/rtos/source/TARGET_CORTEX/rtx5/RTX/Source/rtx_kernel.c: No such file or directory.
(gdb) finish
Run till exit from #0  0x08061682 in osKernelResume (sleep_ticks=14) at ./mbed-os/rtos/source/TARGET_CORTEX/rtx5/RTX/Source/rtx_kernel.c:492
Note: automatically using hardware breakpoints for read-only addresses.
rtos_idle_loop () at ./mbed-os/rtos/source/TARGET_CORTEX/mbed_rtx_idle.cpp:173
173     ./mbed-os/rtos/source/TARGET_CORTEX/mbed_rtx_idle.cpp: No such file or directory.
(gdb) continue
Continuing.`

ubidefeo, 2020-08-25:
hi John

I'm glad you downloaded this right away.
@kittaakos is looking into this, we must have added some regression in the communication between UI and CLI.

As I was telling you a couple days back the debugger didn't get much love this time around, as we had to focus on releasing a more stable editor with broader compatibility, but it's definitely on our list for the next few weeks.
At least now you have access to nightly builds, so if it's something we figure out quickly you'll be able to get a working version faster than 4 months from now :D
 I'll make sure this issue gets updated as we know more.

Thank you :)

jwestmoreland, 2020-08-25:
Hello @ubidefeo ,

Yes, not much love appears to be no love at all actually; ;) - hope this can move forward but there's more ways to debug as we've discussed as well.

That's cool we have access to nightly builds now - thanks for letting us know about that!

And, thanks for looking at the debugger issues as well.

You are more than welcome!

Best Regards,
John W.

kittaakos, 2020-08-25:
I have a quick update: we have identified the breaking change. I built a CLI with a tiny modification locally, dropped it into the IDE and the debugger works.

jwestmoreland, 2020-08-25:
Hello @kittaakos ,

So, will this be in the nightly update or how can I give it a try?

Thanks,
John W.

kittaakos, 2020-08-25:
> will this be in the nightly

Yes, once we fix the issue and very it, we will make it available in the nightly.

> how can I give it a try?

I will keep you posted. Thank you for logging the bug.

jwestmoreland, 2020-08-25:
Hello @kittaakos ,

Thanks for getting to this so quickly!

And, sure, NP - been waiting for the ThreadLib stuff to get stable.

Thanks Again,
John W.

ubidefeo, 2020-08-26:
@jwestmoreland 
you could try the nightly but we're also doing a patch release possibly today

kittaakos, 2020-08-26:
> you could try the nightly

Please note, the nightly won't work until we have another CLI release.

kittaakos, 2020-08-26:
@jwestmoreland, we have a new IDE release with the debugger fix: https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.1

jwestmoreland, 2020-08-27:
Hello @kittaakos ,

OK - thanks for that - and loaded it - just wondering, has anyone debugged anything with it?

From what I can tell, status unchanged from [here.](https://github.com/arduino/ArduinoCore-mbed/issues/19)

Thanks,
John W.

kittaakos, 2020-08-27:
> has anyone debugged anything with it?

Of course.

On Windows and macOS, we verified the basic debugging features on a compiled sketch,
 - such as start debugging,
 - hit breakpoint,
 - step in/out,
 - change local variables (`int`),
 - stop/restart the debug session.

On both Windows and macOS, we used an Arduino Zero board, plus we tried it with CMSIS-DAP debugger on Windows.

> Reading symbols from c:/tmp/demo-web-server/build/arduinobeta.mbed.envie_m7/demo-web-server.ino.elf...
> done.

I do not know if debugging works with the Portenta board. @ubidefeo, can you please help with this?

jwestmoreland, 2020-08-27:
Hello @kittaakos ,

Thanks for the response.

Since this is the arduino-pro-ide 'forum' - thought Portenta as the target was/is implied.

Thanks,
John W.

pnaybour, 2020-08-27:
@jwestmoreland i am not to sure about the support for Portenta board.

jwestmoreland, 2020-08-27:
Hello @pnaybour ,

TheadLib is in the Portenta library source files - so, why would that be the case?

Thanks,
John W.

ubidefeo, 2020-08-28:
@jwestmoreland 
as I was pointing out the other day this release was largely focusing on getting a more stable and usable IDE out the door.
Now we have the platform (nightly builds are a huge part of this new workflow) to accept requests for change and show that we work on features.
Portenta is a big part of this transition, and we're actively investigating its behaviour with CLI and Pro IDE as well as looking into how to offer a working debugger that people actually want and can use

You and other users will be able to see things moving at a faster pace, so please keep on submitting your issues and telling us what doesn't work so we can make a plan to fix it :)

jwestmoreland, 2020-08-28:
Hello @ubidefeo ,

Absolutely, as evidenced [here.](https://github.com/arduino/ArduinoCore-mbed/issues/46)

I hope the nightly builds will have a log as to which bugs could have been fixed or at least are ready to test to see if they've been fixed - by issue number.

Thanks!
John W.

ubidefeo, 2020-08-29:
@jwestmoreland 
nightly builds at the moment do not come with a commit history, but let's look into how we can automate that.
@silvanocerza @kittaakos this one is for you, I guess ⬆️

On Monday I'll check with @facchinm what's left to do to enable debugging via ThreadDebug on Portenta, we already chatted about it a few days ago but need to collect more info to make into a clear development/fix task. I'll notify you as soon as we have something.

Thank you again for being (or trying to be) so active with Portenta and the new IDE, hope it doesn't cause too much frustration as you bring valuable feedback to this project :)

jwestmoreland, 2020-08-29:
Hello @ubidefeo ,

OK - yes, I think a commit history will be really helpful in this stage of things - thanks for that.

Seems like ThreadDebug is really close to being functional and that will be extremely helpful.  I also have a thread [here:](https://github.com/arduino/ArduinoCore-mbed/issues/19) that's the original post I had regarding ThreadLib.

Thanks for discussing with Martino, and sure, I think we all will benefit once this is solid - I really enjoy embedded systems.

Best Regards,
John W.

kittaakos, 2020-09-03:
> I hope the nightly builds will have a log

It's available. Please reference the updated documentation [here](https://github.com/arduino/arduino-pro-ide#nightly-builds). [Here](https://downloads.arduino.cc/arduino-pro-ide/nightly/CHANGELOG.txt) is the direct link to the changelog.

jwestmoreland, 2020-09-03:
Hello Akos,

Thanks for doing this.  Much appreciated.

Regards,
John

kittaakos, 2020-09-11:
To sum up:
 - As I wrote [here](https://github.com/arduino/ArduinoCore-mbed/issues/46#issuecomment-690963037), IDE2 does not do more than calling the debugger.

> From the command line - using what was compiled by the 0.1.0 IDE - it does run: [...]

 -  @jwestmoreland, I do not know how did you end up with that command, something else is defined in the [`platform.txt`](https://github.com/arduino/ArduinoCore-mbed/blob/8911e84b92de6f4555f45b0fbd1dcf4be62f3e36/platform.txt#L165-L174) of the core, but after adjusting the command that worked for you to my macOS env, I got no output:
```
 % /Users/akos.kitta/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin//arm-none-eabi-gdb -ex "set pagination off" --baud 230400 -ex "set target-charset ASCII" -ex "set print pretty on" -ex "set remotelogfile mri.log" -ex "target remote /dev/cu.usbmodem14601" /Users/akos.kitta/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-apple-darwin10 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/akos.kitta/Documents/Arduino/Blink/build/arduino-beta.mbed.envie_m7/Blink.ino.elf...done.
```
 
 - There was another probably important comment [here](https://github.com/arduino/ArduinoCore-mbed/issues/19#issuecomment-660596860) that I do not understand:

> OK - I changed COM11 to COM2 and it appears to have fixed that...on the CLI...

Where did you get this info from? Can you please help, @jwestmoreland? Once I can run gdb from a terminal using the CLI, I can make it work with IDE2. Thank you!

jwestmoreland, 2021-02-24:
@kittaakos ,

Can you let me know what's unclear now - I'm wondering what the actual question(s) is/are that you want me to answer?

Thanks,
John W.

ubidefeo, 2021-02-26:
hi @jwestmoreland 
This is a fairly old thread.
We have since assessed that changes are to be made to the `arduino:mbed` core in order to get the ThreadDebug port to be recognised and used at the start of a debug session.

The issue will remain open until we have this implemented and working :)