arduino/arduino-pro-ide #313: 0.1.0 - Can Upload Just Fine In Bootloader Mode - Thanks!

jwestmoreland, 2020-08-25:
Hello All,

I just wanted to comment that under 0.0.6 I couldn't ever get the Portenta to upload in bootloader mode - and wanted to comment I'm glad that bug appears to be fixed with 0.1.0.

Thanks For That!

Best Regards,
John W.

ubidefeo, 2020-08-25:
hey @jwestmoreland 
you should be able to upload even when not in BL mode.
Please keep'em coming 🙏

jwestmoreland, 2020-08-26:
Ah, yes @ubidefeo - funny.  ;)

It was a nice 'surprise' to me anyway when that worked - I suppose it was logged already someplace as a 'bug' - but running 1.8.13 when 'stuck' in bootloader mode just wasn't that big of a deal except for the arduino-beta arduinobeta thing.

Regards,
John W.

ubidefeo, 2020-08-26:
@jwestmoreland
as we were preparing for this release we gave a bit of priority to Portenta and realised that some weird issues needed addressing.
I'm happy to hear you consider them addressed ;)

closed 2020-08-26 by ubidefeo