arduino/arduino-pro-ide #314: Autocomplete lacks semicolons

Rainbowz4U, 2020-08-25:
When typing commands that normally require a semicolon at the end, the autocomplete is able to guess and fill in the command, but it always lacks a semicolon at the end, causing a compiler error.

**Screenshots**
[image]
[image]

**Desktop:**
 - OS: macOS Catalina
 - Version 0.1.0

kittaakos, 2020-08-25:
Thanks for reporting it, @Rainbowz4U. This is definitely not a bug, but the expected behavior, so I tend to remove the bug label and replace with enhancement.

I have been working with several IDEs, tools, and editors in the past, but the automatic semicolon insertion was never part of the IntelliSense feature. Can you name a tool/IDE that does the semicolon insertion after auto-complete? Does VS Code, for example, with the C++ development extension supports it?

kittaakos, 2020-08-25:
This VS Code extension could help to support this feature request: https://marketplace.visualstudio.com/items?itemName=seunghwan.smartsemicolon

Rainbowz4U, 2020-08-25:
I tested within VS Code, and the IntelliSense in the C++ extension does add in semicolons for you. I was able to have it work with just the C++ extension, and no need for the smart semicolon.
[image]
[image]

kittaakos, 2020-08-25:
> the IntelliSense in the C++ extension does add in semicolons for you.

Thank you for your help! The screenshot you have attached is a template and not a symbol if I am not mistaken. Does it also work if you auto-complete a method? Maybe this [can be configured](https://code.visualstudio.com/docs/editor/intellisense#_customizing-intellisense) without any additional extensions, I am just not yet aware of it.

Rainbowz4U, 2020-08-25:
I tested again, and it has trouble auto-completing methods. It seems to only work for simple commands, as when I type in a method, it only fills in the name without parameters, parenthesis, or the semicolon. The current IntelliSense in the Pro IDE encounters the same issue.

ubidefeo, 2020-08-26:
@Rainbowz4U 
I am using it the way I always have used intellisense:
* type the function name
* get suggestion
* TAB to parameter(s)
* TAB once again to exit the signature and add the semi-colon at the end

maybe because I got used to manually add that extra TAB to place the semicolon or exit the signature in order to chain more commands for one-liners

closed 2021-02-18 by ubidefeo