arduino/arduino-pro-ide #315: Confirm Exit as 'Always' under Preferences Makes IDE Freezes Upon Quitting

the13thparish, 2020-08-25:
Dear all, 

There is a  glitch on the Mac version of the 0.1.0 Pro IDE regarding quitting the program. I only have a Mac to hand, so I can't tell whether this is a general issue, or specific to the Mac version of the Pro 0.1.0 IDE.  Under Preferences, if you have the Confirm Exit option as 'always' the IDE won't close; you have to force quit 'Arduino Pro IDE' on the Activity Monitor (the Mac equivalent to Task Manager).

This on its own is still insufficient, because if you reopen the Pro IDE, you cannot select a board or port at all and the software says that it is offline in the yellow bar at the bottom of the program.

A way around this for now is to make sure to force quit the Arduino Pro IDE as well as ALL instances of 'Arduino Pro IDE helper'; for some reason I have several of these open at any given time and they do not all close when the IDE is forced to close. 

Secondly if you have the same Confirm Exit option as 'always', have a a sketch open want to save a copy of it under a different name, a pop up box asks me 'would you like to quit?'. The program does not quit, but freezes as mentioned above. 

All the best,
Ryan

kittaakos, 2020-08-25:
> this is a general issue

Thank you for reporting this bug, @the13thparish 👍 

It is a generic issue and tracked here: https://github.com/eclipse-theia/theia/issues/8186
We have to fix it for sure. For the time being, as a workaround, I would recommend setting this property to `never` as the auto-save will make sure your sketch content is saved to the file system.

per1234, 2021-02-17:
This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4). I can reproduce it with 0.1.2, but not with 0.1.4, so I think we have good news. Thanks Akos!

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-17 by per1234