arduino/arduino-pro-ide #316: Basic commands underlined as error

OslafEman, 2020-08-26:
Arduino-pro-ide 0.1.0 highlights as error basic commands such as delay, serial, wifi, etc.

**To Reproduce**
Open any code that contains these commands

**Screenshots**
[image]

**Desktop (please complete the following information):**
 Windows 10 2004

ubidefeo, 2020-08-26:
@kittaakos 
something to do with the LS support when ESP is selected?

kittaakos, 2020-08-26:
@OslafEman, which board did you select when you had this error? Have you installed the corsponding core?

OslafEman, 2020-08-26:
Hi @kittaakos, thanks for the prompt reply. I have selected the MH ET LIVE ESP32 MiniKit board and also installed the corresponding core.

[image]

per1234, 2021-02-17:
Hi @OslafEman . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-17 by per1234