arduino/arduino-pro-ide #317: The File - Open Folder option has been removed

OslafEman, 2020-08-26:
In the alpha versions I used the File - Open Folder option. The opened folder could contain in some cases several subfolders with different codes or in other cases different versions of the same code.
In version 0.1.0 that is no longer possible. I think a nice feature of the program has been missed.

Arduino Pro IDE 0.1.0
[image]

Alpha versions
[image]

ubidefeo, 2020-08-26:
hi @OslafEman 
In the new release we reworked the menu system to be more aligned with our legacy Java IDE.
This happened to ease the transition of current users to the new IDE.
Nevertheless this issue you reported is of great use to us because it tells us what an advanced user would like to have as a feature.
Can you confirm you're using the IDE switched to Advanced Mode?
We'll keep it open as feature request :)

per1234, 2020-08-26:
*Merged from https://github.com/arduino/arduino-pro-ide/issues/321 by @per1234*

---
Hello @ubidefeo, thank you for your prompt reply. I confirm that I am using the Advanced Mode of the IDE.

[image]