arduino/arduino-pro-ide #318: Sketch folder name not recognised

jdmack-uk, 2020-08-26:
**Describe the bug**
When I open an existing sketch, say Blink.ino, I get the message:
_The_file "Blink.ino" needs to be inside a sketch folder named as "Blink".
Create this folder, move the file, and continue?_
My Blink.ino **is** in C:\Users\name\Documents\Arduino\Blink
If I click OK, it creates a new folder there, so the sketch is now in C:\Users\name\Documents\Arduino\Blink\Blink - but at least it opens it. If I then close the IDE and start it again, I get the same message! So the Blink.ino gets one \Blink level deeper each time!
For some reason it's not recognising that the file is already in a Blink folder.

**To Reproduce**
1. I installed the 0.1.0 standard zip, then removed it and tried the 20200825 nightly build. Same thing on both.
2. Click on 'Open' button
4. See error

**Expected behavior**
It needs to accept the fact that the folder name is OK

**Desktop (please complete the following information):**
 - OS: Windows 7 PRO 64bit
 - Version: Arduino PRO IDE 0.1.0-nightly.20200825

arduino-cli.exe is on my PATH.
I tried running the IDE as Administrator and nothing changed.
My arduino-cli.yaml has the following line:
  *user: C:\Users\name\Documents\Arduino*
... as well as other stuff which I haven't touched.
The IDE executable is *E:\name\Downloads\arduinoPRO\Arduino Pro IDE.exe* (not on PATH).

Many thanks for your help.

ubidefeo, 2020-08-26:
hi @jdmack-uk 
we used to have this bug in a pre-release but it was fixed.
I cannot reproduce it on my system but I'll ask the team to investigate.
Thank you for reporting 👍

kittaakos, 2020-08-26:
Thank you for reporting the problem, @jdmack-uk. I could reproduce it with `0.1.0` and I can confirm; this issue is the same as #311. We have fixed it yesterday and it is available as the `0.1.0-nightly.20200826` nightly. Could you please get the latest nigthly and try it once more? Thank you!

LumitoLuma, 2020-08-26:
@jdmack-uk The error seems to be fixed with version [0.1.1](https://github.com/arduino/arduino-pro-ide/releases/0.1.1).

jdmack-uk, 2020-08-26:
@kittaakos I have installed Windows 64 v 0.1.1 and can **confirm the issue is fixed**!
Thank you so much for your prompt reply and help!

closed 2020-08-26 by jdmack-uk