arduino/arduino-pro-ide #322: The IDE should handle invalid CLI configuration gracefully

kittaakos, 2020-08-27:
Originally from here: https://github.com/arduino/arduino-pro-ide/issues/310

Start your app with such an invalid config and it hangs:
```yaml
board_manager:
  additional_urls:
  https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
  https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
daemon:
  port: "50051"
directories:
  data: /Users/akos.kitta/Library/Arduino15
  downloads: /Users/akos.kitta/Library/Arduino15/staging
  user: /Users/akos.kitta/Documents/Arduino
logging:
  file: ""
  format: text
  level: info
telemetry:
  addr: :9090
  enabled: true
```

ubidefeo, 2021-02-18:
this has been addressed by @kittaakos and @silvanocerza in one of the latest iterations of command/config binding between IDE and CLI

per1234, 2021-02-18:
@ubidefeo what results do you get if you put invalid YAML like this in your Arduino Pro IDE `arduino-cli.yaml` file?:
```yaml
board_manager:
  additional_urls:
  https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
  https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
```
When I do that, starting Arduino Pro IDE hangs at the splash screen, as reported by Akos.

So it seems to me that this issue has not yet been resolved. Even though it's not so likely to happen now that the Pro IDE's CLI configuration is not so exposed to the user, I still like the idea of making it handle that eventuality gracefully, either by recovering or else communicating about the problem to the user.

ubidefeo, 2021-02-18:
@per1234 
in the case of malformed lists/arrays of additional URLs this should not be an issue because those are filled in using the recently introduced Preferences GUI.
Other settings do require the IDE and the CLI to agree upon