arduino/arduino-pro-ide #323: ar.exe (Windows) cannot be executed

jdmack-uk, 2020-08-28:
**Describe the bug**
When "verifying" a sketch (e.g. Blink.ino) I get the following error:
`"C:\\Users\\myname\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs "C:\\Users\\myname\\AppData\\Local\\Temp\\arduino-sketch-EED4448FFEA2184C6E33501D842E49DE\\core\\core.a" "C:\\Users\\myname\\AppData\\Local\\Temp\\arduino-sketch-EED4448FFEA2184C6E33501D842E49DE\\core\\wiring_pulse.S.o"
Error running c:\users\myname\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: CreateProcess
Compilation error: Error: 2 UNKNOWN: exit status 1`

The command calls `...\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar`, 
but the error comes from `...\7.3.0-atmel3.6.1-arduino7/avr/bin/ar.exe`

When I try and execute ar.exe in its folder, I get "Access is denied":
`C:\Users\myname\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\bin>ar`
`Access is denied.`

I have checked file permissions and they're OK for my username and Administrators.
I am an Administrator on my laptop, but I also tried right-clicking it and "Run as Administrator",  which gives me a message box saying "Windows cannot access the specified device, path or file. You may not have the appropriate permissions to access the item".

**To Reproduce**
1. Remove AppData\Local\Arduino15 folder (for a clean start with the boards/tools)
2. Start Arduino Pro IDE 0.1.1 (either Classic or Advanced mode - the behaviour is the same)
3. Open Blink.ino
4. Open Board Manager
5. Install "Arduino AVR Boards by Arduino" v 1.8.3
6. Select board: Arduino Nano (not connected, so no com port; I also tried the Mega, same results)
7. Make sure Output pane is visible
8. Click "Verify" button
The attached log file contains these instructions and the relevant contents of the Output pane.
[StepsToReproduceIssue.log](https://github.com/arduino/arduino-pro-ide/files/5139259/StepsToReproduceIssue.log)

**Expected behavior**
`avr-gcc-ar.exe` should run or the makefile should call `avc-ar.exe` (see Additional context)

**Desktop (please complete the following information):**
 - OS: [Windows 7 Pro]
 - Version [Arduino Pro IDE 0.1.1] (same happened in 0.1.0 and 0.1.0 nightly-20200825, i.e. all versions of Pro that I have tried)

**Additional context**
1. I have tried both in Classic and Advanced mode (not surprisingly the behaviour is the same!)
2. The legacy (non-Pro) IDE builds the sketch correctly (and I uninstalled it before running Pro, to avoid conflicts - part of the reason for cleaning out the Arduino15 folder).
3. I notice that in `...\7.3.0-atmel3.6.1-arduino7/bin/` there are:

- `avr-gcc-ar.exe (52,736 bytes)` and
- `avr-ar.exe (630,784 bytes)`

If I run `avr-gcc-ar.exe` with no parameters, I get the same error message as above:
`Error running c:\users\myname\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: CreateProcess`

If I run `avr-ar.exe` with no parameters, I get a usage screen:
`C:\Users\myname\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin>.\avr-ar`
`Usage: .\avr-ar [emulation options] [-]{dmpqrstx}[abcDfilMNoPsSTuvV] [--plugin <name>] [member-name] [count] archive-file file...`
`       .\avr-ar -M [<mri-script]`
` commands:`
`  d            - delete file(s) from the archive`
... etc.
If I ask for the version, I get:
`GNU ar (GNU Binutils) 2.26.20160125`
`Copyright (C) 2015 Free Software Foundation, Inc.`
`This program is free software; you may redistribute it under the terms of`
`the GNU General Public License version 3 or (at your option) any later version.`
`This program has absolutely no warranty.`

And, finally, I had saved the avc-ar.exe from the legacy IDE installation for comparison, **and it's the same version as the one in the Pro tools!**

**Many thanks!**

kittaakos, 2020-08-28:
Thank you for the details bug report, @jdmack-uk. I'll look into it.

kittaakos, 2020-08-28:
I could not reproduce the problem; it worked for me from both the IDE and the CLI only. What happens, when you do the same using the CLI from a command line? You can get a CLI from [here](https://github.com/arduino/arduino-cli/releases/latest). My steps were the followings:
```
~/Desktop/arduino-cli.exe version \
&& rm -rf ~/AppData/Local/Arduino15/ \
&& rm -rf ~/Documents/Arduino/ \
&& cat ~/Desktop/Blink/Blink.ino \
&& ~/Desktop/arduino-cli.exe core install arduino:avr \
&& ~/Desktop/arduino-cli.exe compile -b arduino:avr:nano:cpu=atmega328 ~/Desktop/Blink/ -v
```

The output was:
```
arduino-cli.exe Version: 0.12.1 Commit: 48383da
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded MiB    0.00%
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded 0.00%
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
Downloading packages...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 downloaded/ 50.09 MiB   98.11% 00m01s
arduino:avrdude@6.3.0-arduino17 downloaded
arduino:arduinoOTA@1.3.0 downloaded 2.64 MiB   53.24%
arduino:avr@1.8.3 downloaded 4.71 MiB   89.91%
Installing arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7...
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 installed
Installing arduino:avrdude@6.3.0-arduino17...
arduino:avrdude@6.3.0-arduino17 installed
Installing arduino:arduinoOTA@1.3.0...
arduino:arduinoOTA@1.3.0 installed
Installing arduino:avr@1.8.3...
Configuring platform...
arduino:avr@1.8.3 installed
Using board 'nano' from platform in folder: C:\Users\kittaakos\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\kittaakos\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\sketch\\Blink.ino.cpp" -o "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\sketch\\Blink.ino.cpp" -o "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\sketch\\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR [... cut ...]
"C:\\Users\\kittaakos\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\KITTAA~1\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC/Blink.ino.elf"
Sketch uses 924 bytes (3%) of program storage space. Maximum is 30720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
```

Note: I used Git Bash running it. I guess the steps are slightly different from `CMD.exe` but I hope you get the idea.

jdmack-uk, 2020-08-29:
Hi @kittaakos, thank you for your help!
I have followed your CLI command suggestion and the result was the same error.
Here are the details:

I put your command into a script called ardutest, to adapt it to my settings.
arduino-cli is on my path (and I downloaded the latest one from the link you sent).
I also ran this on GitBash as I already had it installed!
I didn't delete the `~/Documents/Arduino` folder because that's where my sketches are.

```
$ cat ardutest
arduino-cli.exe version \
&& rm -rf ~/AppData/Local/Arduino15/ \
&& cat ~/Documents/Arduino/Blink/Blink.ino \
&& arduino-cli.exe core install arduino:avr \
&& arduino-cli.exe compile -b arduino:avr:nano:cpu=atmega328 ~/Documents/Arduino/Blink/ -v
```

I used WinMerge to compare your output to mine so I edited them, changing our usernames to "username" and the sketch ID to yours (34C9C83806E62523E2E314562BD9B9BC)
Here is what I noticed in the comparison:
1. Where it seems to matter (downloading and installing the tools) everything is the same
2. I found it strange that the files were compiled in a different order - but that shouldn't be relevant to the error
3. The problem continues to be avr-gcc-ar.exe calling ar.exe

Interestingly, I tried to run ar.exe in GitBash and it gave me the following error:
```
$ ./ar.exe
sh.exe": ./ar.exe: Bad file number
```
It is a bit better than "Access is Denied" but still not very helpful.

Here is the output of the ardutest script, with the modifications:
```
$ ./ardutest
arduino-cli.exe Version: 0.12.1 Commit: 48383da
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  This example code is in the public domain.
 */

// Pin 13 has an LED connected on most Arduino boards.
// give it a name:
int led = 13;

// the setup routine runs once when you press reset:
void setup() {
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(900);               // wait for a second
}
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 downloaded
Installing builtin:serial-discovery@1.0.0...
builtin:serial-discovery@1.0.0 installed
Updating index: library_index.json downloaded
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
[... 30 lines cut ...]
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino\\wiring_digital.c" -o "C:\\Users\\username\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\core\\wiring_digital.c.o"
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino\\wiring_pulse.c" -o "C:\\Users\\username\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\core\\wiring_pulse.c.o"
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\eightanaloginputs" "C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino\\wiring_shift.c" -o "C:\\Users\\username\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\core\\wiring_shift.c.o"
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\ [... cut ...]
"C:\\Users\\username\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc-ar" rcs "C:\\Users\\username\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\core\\core.a" "C:\\Users\\username\\AppData\\Local\\Temp\\arduino-sketch-34C9C83806E62523E2E314562BD9B9BC\\core\\wiring_pulse.S.o"
**Error running c:\users\username\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\bin\../lib/gcc/avr/7.3.0/../../../../avr/bin/ar.exe: CreateProcess
Error during build: exit status 1**
```
Wouldn't it be an idea to compare our executables (avr-gcc-ar.exe, avr-ar.exe and ar.exe)?

kittaakos, 2020-08-29:
> Interestingly, I tried to run ar.exe in GitBash and it gave me the following error:

Can you please try to run the same from a `CMD.exe`? Maybe it produces a more meaningful error message. Mine was:
```
C:\Users\kittaakos\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\bin>ar.exe
Usage: ar.exe [emulation options] [-]{dmpqrstx}[abcDfilMNoPsSTuvV] [--plugin <name>] [member-name] [count] archive-file file...
       ar.exe -M [<mri-script]
 commands:
  d            - delete file(s) from the archive
  m[ab]        - move file(s) in the archive
  p            - print file(s) found in the archive
  q[f]         - quick append file(s) to the archive
  r[ab][f][u]  - replace existing or insert new file(s) into the archive
  s            - act as ranlib
  t            - display contents of archive
  x[o]         - extract file(s) from the archive
 command specific modifiers:
  [a]          - put file(s) after [member-name]
  [b]          - put file(s) before [member-name] (same as [i])
  [D]          - use zero for timestamps and uids/gids
  [U]          - use actual timestamps and uids/gids (default)
  [N]          - use instance [count] of name
  [f]          - truncate inserted file names
  [P]          - use full path names when matching
  [o]          - preserve original dates
  [u]          - only replace files that are newer than current archive contents
 generic modifiers:
  [c]          - do not warn if the library had to be created
  [s]          - create an archive index (cf. ranlib)
  [S]          - do not build a symbol table
  [T]          - make a thin archive
  [v]          - be verbose
  [V]          - display the version number
  @<file>      - read options from <file>
  --target=BFDNAME - specify the target object format as BFDNAME
 optional:
  --plugin <p> - load the specified plugin
 emulation options:
  No emulation specific options
ar.exe: supported targets: elf32-avr elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex

C:\Users\kittaakos\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\bin>
```

Thank you!

jdmack-uk, 2020-08-30:
To cut a long story short, I have been experimenting and the problem seems to be the name of the executable!
ar.exe is the name of a known virus:
[https://file-intelligence.comodo.com/windows-process-virus-malware/exe/ar](https://file-intelligence.comodo.com/windows-process-virus-malware/exe/ar)
I changed the file's name to bad-ar.exe and it runs. I copied avr-ar.exe (which runs) to ar.exe and it stops running!
So it seems something on my laptop is preventing it from running due to its name.
It also seems that **where** the file is in the filesystem is important:
I have MinGW installed, which also has two ar.exe files:
`/c/MinGW/bin/ar.exe` and
`/c/MinGW/mingw32/bin/ar.exe`
**Both run fine!**
So I copied the first one to the `.../avr/bin/` folder - and it doesn't run!
Then I copied the original (problematic) `ar.exe` to my `Documents` folder and it runs fine! That also explains why the legacy Arduino IDE `ar.exe` runs normally, because it's in it's own folder tree, not under `AppData` (or whichever component of the path is causing the problem).

So what I would like to ask is:
- have you seen this kind of issue before?
- is there a way to install the Arduino tools in a different location?

Many thanks!

kittaakos, 2020-08-31:
> * have you seen this kind of issue before?

Yes. This problem seems to be common and persists a long time ago without a proper solution.

From [here](https://forum.arduino.cc/index.php?topic=529334.msg3608290#msg3608290) and [here](https://forum.arduino.cc/index.php?topic=526735.msg3592633#msg3592633):
> This sort of error could be caused by your antivirus software. Try **TEMPORARILY** disabling your antivirus for a single compilation to see if the problem goes away, then turn the antivirus back on. If the problem doesn't occur with the antivirus off you will need to adjust the settings of your antivirus to whitelist the appropriate file, folder, or process so it doesn't interfere with compilation.

> * is there a way to install the Arduino tools in a different location?

Fortunately, yes. You can tune the `directories.data` and `directories.downloads` folders in the CLI configuration. [Here](https://github.com/arduino/arduino-pro-ide/issues/245#issue-613309368), you can find some documentation on how to change the CLI configuration.

I am going to leave this issue open for greater visibility and further references, but I am afraid there is not much the CLI or the Pro IDE can do about it.

jdmack-uk, 2020-09-01:
Hi @kittaakos,
Wow! The more I hear about the CLI, the more I want to learn about it!
I plan to change the data and downloads locations because I'll put them on another partition with more space than my C: drive.
I totally agree that, concerning the antivirus behaviour, it has become my local problem, so there's nothing to add to this issue.
Thank you so much for all your help!
Best regards,
@jdmack-uk

closed 2021-02-17 by per1234