arduino/arduino-pro-ide #324: 0.1.1 (etc.) console "error" string text too dull to easily read (please make bright red, not dull red)

paschlie, 2020-08-28:
This is really a problem with the non-pro version, as error messages are nearly impossible to read, being displayed as very dull red text; please brighten (as error messages should not be difficult to read).  Thanks.

kittaakos, 2020-08-31:
@paschlie, I can see that the theme causes some frustration on your side, we got the message from your [previous GH issue](https://github.com/arduino/arduino-pro-ide/issues/310) too. Unfortunately, we have [not received a single](https://github.com/arduino/arduino-pro-ide/issues/310#issuecomment-679304706) screenshot depicting the UI/UX problem:

> @paschlie, can you please attach a screenshot to the issue?

Can you please attach a screenshot just to see how bad the situation is?

As a workaround, please use the Arduino theme, that's a light one and the default 😄

paschlie, 2020-08-31:
Using the "Arduino (light)" theme (which uses a dark background console display like (dark) themes but for some reason the "red" in dark themes is brighter; maybe the two different red fonts should be used, a darker one for light backgrounds, and a brighter one for dark backgrounds):

[image]

kittaakos, 2020-08-31:
Thank you for the update, we will keep your feedback in mind when it comes to the theme updates 👍

kittaakos, 2020-12-01:
The foreground color of the font is always white in the _Output_, no matter it was an error or not.

[image]

Closing.

closed 2020-12-01 by kittaakos