arduino/arduino-pro-ide #325: v0.1.0, v0.1.1 Editor support may be inaccurate because the header Header.h was not found

Bad-ptr, 2020-09-01:
I'm trying it on windows.
The header is in the same "sketch" folder as the .ino file.

v0.0.6 works fine(but only .ino files syntactically highlighted).

kittaakos, 2020-09-01:
@Bad-ptr, thank you for reporting the issue. Please share the installed core, the selected board, and possibly the sketch folder content (the sketch file and the header) with us.

Bad-ptr, 2020-09-01:
>  installed core
Just downloaded and unpacked the archives from the [releases page](https://github.com/arduino/arduino-pro-ide/releases).

Bad-ptr, 2020-09-01:
>  installed core

Just downloaded and unpacked the archives from the [releases page](https://github.com/arduino/arduino-pro-ide/releases). And installed the `Arduino AVR boards` components.

> selected board

Arduino Genuine Uno

> possibly the sketch folder content (the sketch file and the header)

Just created a new sketch, added `Header.h` file with a declarations in it: `namespace MyTest { void test(){return;} }`,  and included it in .ino with `#include "Header.h"`.

After that the IDE can not autocomplete `MyTest::test` when I edit .ino file. It underlines the word `MyTest` like it is error (symbol not found or something).
If I restart the IDE it gives me that warning: "Editor support may be inaccurate because the header Header.h was not found..."

The sketch folder is in the root of disk D:\ the name of the sketch is autogenerated `sketch_sep1a`.

eirannejad, 2020-09-03:
I installed the Adafruit Servo package using the IDE but `clangd` is not configured to find the package headers so `#include <Adafruit_PWMServoDriver.h>` does not find the header. This seems to be a language server issue only. The compiler works just fine

kittaakos, 2020-09-03:
Related: https://github.com/arduino/arduino-pro-ide/issues/149

eirannejad, 2020-09-05:
I managed to solve this by revising the `user` path in the `arduino-cli` config file to
```yaml
user: /Users/ein/Library/Arduino15/packages/arduino/hardware/avr/1.8.3
```

This forces the `arduino-cli` to install libraries into this directory and fixes the language server issue not being able to find them

ubidefeo, 2020-09-05:
@eirannejad 
although this works, if you upgrade the AVR core all your libraries might be lost and you'll be back at square one.
libraries should be installed in their own path in the sketch folder to make them permanent.

the best solution will come once our Language Server will be fixed and all the headers will be loaded properly for code analysis

AHardReset, 2020-10-05:
> @eirannejad
> although this works, if you upgrade the AVR core all your libraries might be lost and you'll be back at square one.
> libraries should be installed in their own path in the sketch folder [...]

This means that it will be fixed later?

I have the same issue Specs:
Arduino Pro IDe 0.1.2-nightly.20201004
Windows10
Arduino Nano. Atmega 328P (Old Botloader)

With a header in the same directory the problem described here appears.
[image]

[image]

Even with libraries from the "Library Manager"
the classes are not recognized

[image]

Let me know if I can do something to help!

ubidefeo, 2020-10-05:
@AHardReset 
we've been busy with other tasks, but work on the Language Server is going to happen in the next two weeks, so we plan on finding a solution for this.
a bit of patience, we have a lot of things to do before we go out of Alpha :)

Dave528, 2020-11-11:
I'm having the same issue with the language server telling me that "".

Dave528, 2020-11-11:
I'm having the same issue with a header in the root folder of my project not being found.  The language server does seem to be the problem since the project compiles just fine.
[image]
The message is always "Editor support may be inaccurate because the header "tft_colors.h" was not found.  If it is part of a library, use the Library Manager to install it."  This file is opened by the IDE at startup, and appears in the file list tree.  It is also the last file opened every time the IDE is started. (This may be where the issue occurs, with the language server not scanning the last file opened.)

kittaakos, 2020-11-11:
> The language server does seem to be the problem since the project compiles just fine.

The language server is not involved in the compilation process. The fact that it compiles without a problem but you still see the errors in the editor indicates that this problem is LS-related.

> **(This may be where the issue occurs, with the language server not scanning the last file opened.)**

No. It's unrelated.

jpmeijers, 2020-12-10:
I'm also experiencing this issue using the current Nightly build.

Header files in the same directory as the main .ino file is not found and the "Editor support may be inaccurate because the header XX.h was not found." error is displayed.

I'm also facing an issue with sources that are located under the `src/` subdirectory. It compiles fine, but the IDE does not find these and the references are underlined in red.

ubidefeo, 2020-12-10:
@jpmeijers 
this is going to be a bit of an issue until our work on the language server is complete.
it is unfortunately taking a bit longer than expected but we're very close to getting these errors lifted

Dave528, 2020-12-11:
@ubidefeo 
Kudo's to those of you that are doing this work.  I (probably all of us using the Pro IDE) really appreciate your hard work.
Is the source code available for the Pro IDE?  I haven't been able to locate it.

ubidefeo, 2021-02-18:
@Bad-ptr 
I have just tested this in the Alpha 0.1.4 and it works properly,
hence this will be closed.
It might be one of those glitches the Language Server has here and there, but we're ironing them out.

cc @per1234 

[image]
[image]
[image]

closed 2021-02-18 by ubidefeo