arduino/arduino-pro-ide #326: arduino-cli daemon issues on Arch Linux

groowyCZ, 2020-09-01:
**Describe the bug**
IDE isn't able to connect to the CLI daemon.

**To Reproduce**
Open Arduino Pro IDE

**Expected behavior**
Arduino Pro IDE connects to CLI daemon

**Log**
```
groowy@asus ~> /opt/arduino-pro-ide/Arduino\ Pro\ IDE-v0.0.6-linux-x64/arduino-pro-ide
root WARN please install @theia/electron@0.17.0-next.f5433ece as a runtime dependency
root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/plugins does not exist.
root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/extensions does not exist.
root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/plugins does not exist.
root INFO Configuring to accept webviews on '.+.webview..+' hostname.
root INFO Theia app listening on http://localhost:37437.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/plugins/vscode-yaml
root INFO Resolved "vscode-yaml" to a VS Code extension "vscode-yaml@0.7.2" with engines: { vscode: '^1.31.0' }
config INFO Started watching the Arduino CLI configuration: '/home/groowy/.arduinoProIDE/arduino-cli.yaml'.
root INFO Deploying backend plugin "vscode-yaml@0.7.2" from "/opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/plugins/vscode-yaml/extension/out/src/extension"
root INFO Deploy plugins list took: 34.3 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 71.5 ms
daemon INFO Starting daemon from /usr/bin/arduino-cli...
daemon INFO Using config file: /home/groowy/.arduinoProIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.12.0-arch
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
root INFO Git askpass helper is listening on http://127.0.0.1:42913.
root INFO Using Git [2.28.0] from the PATH. (/usr/bin/git)
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/touch
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @typefox/monaco-editor-core
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-debounce
root WARN Failed to resolve module: @theia/electron
root WARN Failed to resolve module: @theia/messages
root WARN Failed to resolve module: @types/diff
root WARN Failed to resolve module: @types/connect
root WARN Failed to resolve module: @types/mime
root WARN Failed to resolve module: @types/serve-static
root WARN Failed to resolve module: @types/request
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @types/dateformat
root WARN Failed to resolve module: @types/deepmerge
root WARN Failed to resolve module: @types/glob
root WARN Failed to resolve module: @types/google-protobuf
root WARN Failed to resolve module: @types/js-yaml
root WARN Failed to resolve module: @types/ps-tree
root WARN Failed to resolve module: @types/react-select
root WARN Failed to resolve module: @types/sinon
[... 31 lines cut ...]
root INFO Changed application state from 'attached_shell' to 'initialized_layout'.
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba][redhat.vscode-yaml]: Loaded contributions.
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Load contributions of 1 plugin took: 32.3 ms
root INFO Starting language server /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/arduino-ide-extension/build/arduino-language-server -clangd /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd -cli /usr/bin/arduino-cli -fqbn arduino:avr:nano -board-name "Arduino Nano"
root INFO [hosted-plugin: 42252] PLUGIN_HOST(42252): PluginManagerExtImpl/init()
root INFO [hosted-plugin: 42252] PLUGIN_HOST(42252): initializing(vscode-yaml@0.7.2 with /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/@theia/plugin-ext-vscode/lib/node/plugin-vscode-init.js)
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba][redhat.vscode-yaml]: Started plugin.
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Start of 1 plugin took: 537.1 ms
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Sync of 0 plugins took: 2.4 ms
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Load contributions of 0 plugins took: 0.1 ms
root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Start of 0 plugins took: 8.3 ms
root ERROR Arduino: 2020/09/01 18:12:35 Command error: /usr/bin/arduino-cli exit status 1
2020/09/01 18:12:35 ------------------------------BEGIN STDERR
 Error during build: unable to find a sketch file in directory /tmp/ino2cpp-686848847
 ------------------------------END STDERR

root INFO Changed application state from 'initialized_layout' to 'ready'.
```

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Kernel version: 5.8.5-arch1-1

**Additional context**
I tried completely reinstalling the Arduino Pro IDE from the AUR provided by `yay` (version 0.0.6-1) which didn't result in any change of behavior and I'm suspecting that the daemon is stopped by the error listed in the log but I'm not sure what is causing it because when I list the directory `/tmp/ino2cpp-686848847` it returns `IR_test.ino  IR_test.ino.cpp` which leads me to believe that the arduino-cli wants to access the files before they are created. Feel free to correct me because I don't have any idea what's going on and more importantly why is it going on.

kittaakos, 2020-09-04:
I do not see any daemon-related errors in the logs. Can you please try with the [latest release](https://github.com/arduino/arduino-pro-ide/releases/latest)? Thank you!

Master811129, 2020-09-04:
> **Describe the bug**
> IDE isn't able to connect to the CLI daemon.
> 
> **To Reproduce**
> Open Arduino Pro IDE
> 
> **Expected behavior**
> Arduino Pro IDE connects to CLI daemon
> 
> **Log**
> 
> ```
> groowy@asus ~> /opt/arduino-pro-ide/Arduino\ Pro\ IDE-v0.0.6-linux-x64/arduino-pro-ide
> root WARN please install @theia/electron@0.17.0-next.f5433ece as a runtime dependency
> root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/plugins does not exist.
> root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/extensions does not exist.
> root WARN The directory referenced by local-dir:/home/groowy/.arduinoProIDE/plugins does not exist.
> root INFO Configuring to accept webviews on '.+.webview..+' hostname.
> root INFO Theia app listening on http://localhost:37437.
> root INFO PluginTheiaDirectoryHandler: accepting plugin with path /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/plugins/vscode-yaml
> root INFO Resolved "vscode-yaml" to a VS Code extension "vscode-yaml@0.7.2" with engines: { vscode: '^1.31.0' }
> config INFO Started watching the Arduino CLI configuration: '/home/groowy/.arduinoProIDE/arduino-cli.yaml'.
> root INFO Deploying backend plugin "vscode-yaml@0.7.2" from "/opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/plugins/vscode-yaml/extension/out/src/extension"
> root INFO Deploy plugins list took: 34.3 ms
> root WARN Backend ConfigServiceImpl.onStart is slow, took: 71.5 ms
> daemon INFO Starting daemon from /usr/bin/arduino-cli...
> daemon INFO Using config file: /home/groowy/.arduinoProIDE/arduino-cli.yaml
> daemon INFO arduino-cli version 0.12.0-arch
> daemon INFO Setting up Prometheus telemetry on :9090/metrics
> daemon INFO Starting daemon on TCP address 127.0.0.1:50051
> daemon INFO Daemon is now listening on 127.0.0.1:50051...
> root INFO Git askpass helper is listening on http://127.0.0.1:42913.
> root INFO Using Git [2.28.0] from the PATH. (/usr/bin/git)
> root WARN Failed to resolve module: @types/body-parser
> root WARN Failed to resolve module: @types/cookie
> root WARN Failed to resolve module: @types/express
> root WARN Failed to resolve module: @types/fs-extra
> root WARN Failed to resolve module: @types/lodash.debounce
> root WARN Failed to resolve module: @types/lodash.throttle
> root WARN Failed to resolve module: @types/react
> root WARN Failed to resolve module: @types/react-dom
> root WARN Failed to resolve module: @types/react-virtualized
> root WARN Failed to resolve module: @types/route-parser
> root WARN Failed to resolve module: @types/ws
> root WARN Failed to resolve module: @types/yargs
> root WARN Failed to resolve module: @theia/editor
> root WARN Failed to resolve module: @types/rimraf
> root WARN Failed to resolve module: @types/tar-fs
> root WARN Failed to resolve module: @types/touch
> root WARN Failed to resolve module: @types/uuid
> root WARN Failed to resolve module: @typefox/monaco-editor-core
> root WARN Failed to resolve module: @theia/monaco
> root WARN Failed to resolve module: @theia/preferences
> root WARN Failed to resolve module: @theia/navigator
> root WARN Failed to resolve module: @types/p-debounce
> root WARN Failed to resolve module: @theia/electron
> root WARN Failed to resolve module: @theia/messages
> root WARN Failed to resolve module: @types/diff
> root WARN Failed to resolve module: @types/connect
> root WARN Failed to resolve module: @types/mime
> root WARN Failed to resolve module: @types/serve-static
> root WARN Failed to resolve module: @types/request
> root WARN Failed to resolve module: @types/p-queue
> root WARN Failed to resolve module: @types/dateformat
> root WARN Failed to resolve module: @types/deepmerge
> root WARN Failed to resolve module: @types/glob
> root WARN Failed to resolve module: @types/google-protobuf
> root WARN Failed to resolve module: @types/js-yaml
> root WARN Failed to resolve module: @types/ps-tree
> root WARN Failed to resolve module: @types/react-select
[... 35 lines cut ...]
> root INFO Starting language server /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/arduino-ide-extension/build/arduino-language-server -clangd /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/arduino-ide-extension/build/bin/clangd -cli /usr/bin/arduino-cli -fqbn arduino:avr:nano -board-name "Arduino Nano"
> root INFO [hosted-plugin: 42252] PLUGIN_HOST(42252): PluginManagerExtImpl/init()
> root INFO [hosted-plugin: 42252] PLUGIN_HOST(42252): initializing(vscode-yaml@0.7.2 with /opt/arduino-pro-ide/Arduino Pro IDE-v0.0.6-linux-x64/resources/app/node_modules/@theia/plugin-ext-vscode/lib/node/plugin-vscode-init.js)
> root INFO [873a2110-de56-4e5f-b683-7becb7a283ba][redhat.vscode-yaml]: Started plugin.
> root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Start of 1 plugin took: 537.1 ms
> root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Sync of 0 plugins took: 2.4 ms
> root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Load contributions of 0 plugins took: 0.1 ms
> root INFO [873a2110-de56-4e5f-b683-7becb7a283ba] Start of 0 plugins took: 8.3 ms
> root ERROR Arduino: 2020/09/01 18:12:35 Command error: /usr/bin/arduino-cli exit status 1
> 2020/09/01 18:12:35 ------------------------------BEGIN STDERR
>  Error during build: unable to find a sketch file in directory /tmp/ino2cpp-686848847
>  ------------------------------END STDERR
> 
[... cut ...]
> I tried completely reinstalling the Arduino Pro IDE from the AUR provided by `yay` (arduino-pro-ide version 0.0.6-1) which didn't result in any change of behavior and I'm suspecting that the daemon is stopped by the error listed in the log but I'm not sure what is causing it because when I list the directory `/tmp/ino2cpp-686848847` it returns `IR_test.ino IR_test.ino.cpp` which leads me to believe that the arduino-cli wants to access the files before they are created. Feel free to correct me because I don't have any idea what's going on and more importantly why is it going on.

I have this problem too but in ubuntu distro in 0.1.1 version

kittaakos, 2020-09-04:
@Master811129, thanks for the heads-up; it seems you have a different issue. Please try to change the execution rights of the bundled CLI with:
```
chmod +x /path/to/your/arduino-pro-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli
```
Let us know how it went.

kittaakos, 2020-09-04:
@groowyCZ, I am sure the problem is that you're are using the previous version of the IDE and a newer version of the CLI from your `$PATH`.

From https://github.com/arduino/arduino-pro-ide/issues/326#issue-690284057:
```
root WARN Backend ConfigServiceImpl.onStart is slow, took: 71.5 ms
daemon INFO Starting daemon from /usr/bin/arduino-cli...
daemon INFO Using config file: /home/groowy/.arduinoProIDE/arduino-cli.yaml
```

They're incompatible with each other; the [gRPC APIs can still change](https://github.com/arduino/arduino-pro-ide/issues/327#issuecomment-685411680). We removed the CLI support from the `$PATH`. More info about this is [here](https://github.com/arduino/arduino-pro-ide/pull/319) and [here](https://github.com/arduino/arduino-pro-ide/issues/142). Please get the latest version and try it. Thanks!

Master811129, 2020-09-05:
> @Master811129, thanks for the heads-up; it seems you have a different issue. Please try to change the execution rights of the bundled CLI with:
> 
> ```
> chmod +x /path/to/your/arduino-pro-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli
> ```
> 
> Let us know how it went.

Thank you now it can run the daemon

kittaakos, 2020-09-05:
> Thank you now it can run the daemon

Thanks for the update! How did you unpack the zip? I have to check whether we lost the execution rights on Linux.

Master811129, 2020-09-05:
> > Thank you now it can run the daemon
> 
> Thanks for the update! How did you unpack the zip? I have to check whether we lost the execution rights on Linux.

i unpacked it with Archive Manager(default app) and also Winrar (using wine) but I saw the same behavior

Master811129, 2020-09-06:
> @kittaakos I have this problem too but on ubuntu distro in 0.1.1 version.
> The output of the terminal:
> [image]

@kittaakos could you please tell me how can i fix that issues.

Master811129, 2020-09-06:
> > @Master811129, thanks for the heads-up; it seems you have a different issue. Please try to change the execution rights of the bundled CLI with:
> > ```
> > chmod +x /path/to/your/arduino-pro-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli
> > ```
> > 
> > 
> > Let us know how it went.
> 
> Thank you now it can run the daemon
> but with this errors:
> 
> [image]

@kittaakos could you please tell me how can i fix that issues.

kittaakos, 2020-09-06:
> could you please tell me how can i fix that issues.

I think this problem is the same as https://github.com/arduino/arduino-pro-ide/issues/213#issuecomment-594643599.
Please use a terminal to `unzip` the IDe. Otherwise, the symlinks are not preserved.

Let us know if it did not help.

kittaakos, 2020-09-07:
> i unpacked it with the Archive Manager(default app) and also Winrar (using wine) but I saw the same behavior.

I have checked the execution rights, and they seem correct.

```
kittaakos@ubuntu:~/Downloads$ ls -al
total 208748
drwxr-xr-x  2 kittaakos kittaakos      4096 Sep  7 02:18 .
drwxr-xr-x 26 kittaakos kittaakos      4096 Sep  7 02:17 ..
-rw-rw-r--  1 kittaakos kittaakos 213743252 Sep  7 02:18 arduino-pro-ide_nightly-20200907_Linux_64bit.zip
kittaakos@ubuntu:~/Downloads$ unzip -qq arduino-pro-ide_nightly-20200907_Linux_64bit.zip 
kittaakos@ubuntu:~/Downloads$ ./arduino-pro-ide_nightly-20200907_Linux_64bit/arduino-pro-ide 
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
root WARN please install @theia/electron@1.5.0-next.09ef844b as a runtime dependency
root INFO Configuration directory URI: 'file:///home/kittaakos/.theia'
root INFO Configuring to accept webviews on '.+.webview..+' hostname.
root WARN The directory referenced by local-dir:/home/kittaakos/.arduinoProIDE/plugins does not exist.
root WARN The directory referenced by local-dir:/home/kittaakos/.arduinoProIDE/extensions does not exist.
root WARN The directory referenced by local-dir:/home/kittaakos/.arduinoProIDE/plugins does not exist.
root INFO Theia app listening on http://localhost:37253.
config INFO Started watching the Arduino CLI configuration: '/home/kittaakos/.arduinoProIDE/arduino-cli.yaml'.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /home/kittaakos/Downloads/arduino-pro-ide_nightly-20200907_Linux_64bit/resources/app/plugins/vscode-builtin-cpp
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.44.2" with engines: { vscode: '*' }
root INFO Deploying backend plugin "cpp@1.44.2" from "/home/kittaakos/Downloads/arduino-pro-ide_nightly-20200907_Linux_64bit/resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploy plugins list took: 111.5 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 135.9 ms
daemon INFO Starting daemon from /home/kittaakos/Downloads/arduino-pro-ide_nightly-20200907_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Using config file: /home/kittaakos/.arduinoProIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.12.1
daemon INFO Setting up Prometheus telemetry on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:42965.
root INFO Using Git [2.17.1] from the PATH. (/usr/bin/git)
daemon INFO Checking signature [index: /home/kittaakos/.arduino15/package_index.json, signaturefile: /home/kittaakos/.arduino15/package_index.json.sig, trusted: true]
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /home/kittaakos/.arduino15/packages
daemon INFO Loading package arduino from: /home/kittaakos/.arduino15/packages/arduino/hardware
daemon INFO Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.3",Latest:"1.8.3",Name:"Arduino AVR Boards"}]
daemon INFO Checking existence of 'tools' path: /home/kittaakos/.arduino15/packages/arduino/tools
daemon INFO Loading tools from dir: /home/kittaakos/.arduino15/packages/arduino/tools
daemon INFO Loaded tool [tool: Version: 1.3.0 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
daemon INFO Loaded tool [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
daemon INFO Loaded tool [tool: Version: 7.3.0-atmel3.6.1-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
daemon INFO Loaded tool [tool: Version: 6.3.0-arduino17 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin12, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO Loaded tool [tool: Version: 1.9.1-arduino1 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
daemon INFO Loaded tool [tool: Version: 0.9.0-arduino2 Systems: [OS: i386-apple-darwin11, OS: arm-linux-gnueabihf, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO Loaded tool [tool: Version: 0.10.0-arduino13 Systems: [OS: aarch64-linux-gnu, OS: arm-linux-gnueabihf, OS: i686-linux-gnu, OS: i686-mingw32, OS: i386-apple-darwin11, OS: x86_64-linux-gnu]]
daemon INFO Loading package arduino-beta from: /home/kittaakos/.arduino15/packages/arduino-beta/hardware
daemon INFO Loaded platform [platform: {ID:"arduino-beta:mbed",Installed:"1.2.2",Latest:"1.2.2",Name:"Arduino mbed-enabled Boards"}]
daemon INFO Loading package builtin from: /home/kittaakos/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/kittaakos/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/kittaakos/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.0.0]
daemon INFO Adding libraries dir [dir: /home/kittaakos/Arduino/libraries, location: user]
[... cut ...]
-rwxr-xr-x 1 kittaakos kittaakos  3811781 Sep  4 04:10 arduino-language-server
drwxr-xr-x 2 kittaakos kittaakos     4096 Sep  7  2020 bin
drwxr-xr-x 2 kittaakos kittaakos     4096 Sep  7  2020 lib
-rwxr-xr-x 1 kittaakos kittaakos    15140 Jan 10  2020 LICENSE.TXT
```

Please use `unzip` so that the symlinks and the execution rights are also preserved.

I believe all questions have been addressed here. I am going to close this issue. Please let me know if I have overlooked something, and will reopen this or create a follow-up. Thanks!

closed 2020-09-07 by kittaakos

Master811129, 2020-09-07:
@kittaakos `unzip`command fixed the Issues successfully.

kittaakos, 2020-09-07:
> `unzip` command fixed the Issues successfully.

Thank you for getting back us with the update, @Master811129! I truly appreciate your help with this project. Please feel free to name me on any issues.