arduino/arduino-pro-ide #327: Location of CLI config file

jdmack-uk, 2020-09-02:
**Is your feature request related to a problem? Please describe.**
This doesn't describe a problem, it suggests an improvement.

This discussion is based on the Windows environment, but it is probably equally applicable to Linux and Mac.

The CLI config documentation, found [here](https://arduino.github.io/arduino-cli/latest/configuration/#locations), describes where CLI looks for a config file:
```
1. Location specified by the --config-file command line flag
2. Current working directory
3. Any parent directory of the current working directory (more immediate parents having higher precedence)
4. Arduino CLI data directory (as configured by directories.data)
```
However, when one opens the CLI config in the IDE (File->Open CLI configuration) it creates/opens `C:\Users\myname\.arduinoProIDE\arduino-cli.yaml` - and it works fine **in the IDE**.

The discomfort is caused by the fact that if the user would like to use the CLI from the command line, it (the CLI) **doesn't** find the above config file without extra setup work.

**Describe the solution you'd like**
It would seem more user-friendly (especially to beginners, myself included) if `C:\Users\myname\.arduinoProIDE` were in the list of places the CLI looks for its config (probably best as the last element in the list). That way it would be straightforward to have a single config file for both the IDE and the CLI itself.

**Describe alternatives you've considered**
It sounds philosophically bad for the CLI to look for the file in an IDE-related folder, so a slightly more complex (and versatile) solution would be this:
- there is a default location for the CLI's config (e.g. `C:\Users\myname\.arduino-cli` - that's nice because it's similar to the IDE folder)
- the IDE looks for the CLI config in `C:\Users\myname\.arduinoProIDE` **first** and, if not found, uses the one in the default CLI location
- if the IDE is using the default CLI config and changes are made to it (from within the IDE), the IDE asks "Do you want to change the default file or create a specific one for the IDE?"
- if an IDE-specific one is requested, that is put in `C:\Users\myname\.arduinoProIDE` and, from then on, the IDE will find it first and use it
- obviously if the option is to change the default one, the IDE continues to use the default
- there would need to be a way to return to using the default (the simple way to do that being to delete the file in `C:\Users\myname\.arduinoProIDE`)

**Additional context**
This all started because I changed the directories.data and directories.downloads entries in the config file. I edited it in the IDE, then tried to install boards and compile my sketch using the CLI. The downloads and installations all went to their default locations, as if I hadn't changed the configuration!
I was in a situation where I could type the config file and it had my desired settings, but when I did `arduino-cli config dump`, I got the default values. It was only when I used the `--verbose flag` that I discovered that it was using the default values because it couldn't find **any** config file!
This sounds like a situation others might run into.

kittaakos, 2020-09-02:
> However, when one opens the CLI config in the IDE (File->Open CLI configuration) it creates/opens `C:\Users\myname\.arduinoProIDE\arduino-cli.yaml` - and it works fine **in the IDE**.

Yes, this is the expected behavior. The IDE starts the CLI with `arduino-cli daemon --config-file ~/.arduinoProIDE/arduino-cli.yaml` arguments. (This is the simplified version of  the args.) So no CLI config lookup is done.

> The discomfort is caused by the fact that if the user would like to use the CLI from the command line, it (the CLI) **doesn't** find the above config file without extra setup work.

I can understand the discomfort, but we do not yet want to let any tools and apps to share the same CLI config. Although the CLI is already widely used, it still has a `0.x.y` version. The API can change from one day to the other. That's the main reason the IDE comes with its own bundled CLI. The version of the CLI is pinned. We had to pin the CLI version because the IDE does not execute CLI commands but communicates with the CLI via a gRPC API generated from a [bunch of `.proto` files](https://github.com/arduino/arduino-cli/tree/master/rpc). (Some additional info is available [here](https://github.com/arduino/arduino-pro-ide/issues/142).)

> It would seem more user-friendly (especially to beginners, myself included) if `C:\Users\myname\.arduinoProIDE` were in the list of places the CLI looks for its config

> It sounds philosophically bad for the CLI to look for the file in an IDE-related folder,

👍 

> * there is a default location for the CLI's config (e.g. `C:\Users\myname\.arduino-cli` - that's nice because it's similar to the IDE folder)
> * the IDE looks for the CLI config in [...]

Thank you for providing the detailed steps for an enhanced config lookup logic. As far as I know, and @ubidefeo can correct me if I am wrong, we are planning to change the [current lookup logic](https://arduino.github.io/arduino-cli/latest/configuration/#locations), maybe we will unify the config location among the apps and tools but not before having a more stable CLI API.

jdmack-uk, 2020-09-02:
Thank you, @kittaakos, for your detailed explanation!

kittaakos, 2020-09-02:
Let's leave this issue open; your additional user requirements are completely valid.

ubidefeo, 2021-02-18:
@jdmack-uk 
this reply comes as we groom our backlog.
The IDE's internal CLI is something we would like to keep hidden away for now in light of the fact that we pin a specific release in order not to introduce any breaking changes, and sometimes not to break the public CLI we might work on experimental features that are specifically related to gRPC interaction between the IDE and the CLI.

If you want to use the CLI I advice you to install it as a standalone command line tool.
This will still let you share the installed platforms and tools with the IDE but you won't risk something we've already seen happen during development, which is that introducing special configuration parameters might break the IDE because it cannot validate them.

This issue is therefore closed, but feel free to get back to us if you're not ok with it.

Thank you
ubi

closed 2021-02-18 by ubidefeo