arduino/arduino-pro-ide #328: macOS: debugger problem v0.1.1

newweaver, 2020-09-02:
When starting the debugger the following error message pops up:

Error during Debug: Cannot get command line for tool: cannot get programmer tool: undefined 'debug.tool' property

per1234, 2020-09-02:
Hi @newweaver. Which board do you have selected?

newweaver, 2020-09-02:
Hi @per1234 

I selected an Arduino Uno

per1234, 2020-09-02:
Although the Arduino Pro IDE provides the general debugging capability for Arduino sketches, each boards platform must also be configured to enable this.

As previously discussed in https://github.com/arduino/arduino-pro-ide/issues/219, the boards platform of the Arduino Uno doesn't have support for debugging in the Arduino Pro IDE. This is the cause of the error you got.

As far as I know, the Arduino SAMD Boards platform is the only one that currently has debugging support. The[ Arduino Zero](https://store.arduino.cc/arduino-zero) is especially convenient  to use with the Arduino Pro IDE's debugger because the Zero has a build in debugger, so all you need to do is just plug in the USB cable like normal and you're ready to debug! The [Arduino MKR1000](https://store.arduino.cc/arduino-mkr1000-wifi) is also a good board for using with the debugger because it has the standardized 2x5 0.05" pitch SWD header that allows it to easily be connected to any standard debug probe.

closed 2020-09-02 by per1234