arduino/arduino-pro-ide #330: Bluescreen on Windows 10

Blubbaa, 2020-09-02:
**Describe the bug**
I just had a bluescreen on Windows 10 in the exact moment of closing Pro IDE. The actual error was `WDF_VIOLATION (0x0000010D)`. The application was opened in the background for a longer time and i disconnected my ESP32 earlier, so it was probably still searching for the COM device (i.e. the annoying popup message).

**To Reproduce**
Could not reproduce it so far.

**Expected behavior**
No bluescreen when closing the program :)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 19041.450

If you need any additional information I could try to analyze the memory dump.

kittaakos, 2020-09-03:
> If you need any additional information I could try to analyze the memory dump.

Yes, please. Maybe it's a driver issue: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x10d---wdf-violation

Did you install any specific libraries or 3rd part cores? Does the same issue happen with the (Java-based) [Arduino IDE](https://www.arduino.cc/en/Main/Software)?

Blubbaa, 2020-09-03:
I never had any issues with the Java Arduino IDE. I just installed the Pro version and did not change my installed libraries. The only thing I did was installing the ESP32 via the board manager.

Heres the analysis with Windbg:
```

*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

WDF_VIOLATION (10d)
The Kernel-Mode Driver Framework was notified that Windows detected an error
in a framework-based driver. In general, the dump file will yield additional
information about the driver that caused this bug check.
Arguments:
Arg1: 0000000000000005, A framework object handle of the incorrect type was passed to
	a framework object method.
Arg2: 0000000000000000, The handle value passed in.
Arg3: 0000000000001018, Reserved.
Arg4: ffff8b08875ec810, Reserved.

Debugging Details:
------------------

KEY_VALUES_STRING: 1

    Key  : Analysis.CPU.mSec
    Value: 4827

    Key  : Analysis.DebugAnalysisProvider.CPP
    Value: Create: 8007007e on PC-JONAS

    Key  : Analysis.DebugData
    Value: CreateObject

    Key  : Analysis.DebugModel
    Value: CreateObject

    Key  : Analysis.Elapsed.mSec
    Value: 8849

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 94

    Key  : Analysis.System
    Value: CreateObject

    Key  : WER.OS.Branch
    Value: vb_release

    Key  : WER.OS.Timestamp
    Value: 2019-12-06T14:06:00Z

    Key  : WER.OS.Version
    Value: 10.0.19041.1

ADDITIONAL_XML: 1

OS_BUILD_LAYERS: 1

BUGCHECK_CODE:  10d

BUGCHECK_P1: 5

BUGCHECK_P2: 0

BUGCHECK_P3: 1018

BUGCHECK_P4: ffff8b08875ec810

BLACKBOXBSD: 1 (!blackboxbsd)

BLACKBOXNTFS: 1 (!blackboxntfs)

BLACKBOXPNP: 1 (!blackboxpnp)

BLACKBOXWINLOGON: 1

PROCESS_NAME:  arduino-cli.exe

STACK_TEXT:  
ffff9f83`1d7af088 fffff800`537f8920     : 00000000`0000010d 00000000`00000005 00000000`00000000 00000000`00001018 : nt!KeBugCheckEx
ffff9f83`1d7af090 fffff800`537beca5     : ffff8b08`88b13670 ffffa10e`00000580 fffff800`53838460 ffff8b08`88b13670 : Wdf01000!FxVerifierBugCheckWorker+0x24 [minkernel\wdf\framework\shared\object\fxverifierbugcheck.cpp @ 68] 
ffff9f83`1d7af0d0 fffff800`ad345790     : 00000000`00000000 00000000`00000000 ffff8b08`90df9020 ffff9f83`1d7af1a9 : Wdf01000!imp_WdfFileObjectGetDevice+0x1b885 [minkernel\wdf\framework\shared\core\fxfileobjectapi.cpp @ 173] 
ffff9f83`1d7af110 fffff800`537c4303     : ffff8b08`875ec810 00000000`00000000 00000000`00000000 00000000`00000000 : silabser+0x5790
ffff9f83`1d7af140 fffff800`537aa867     : ffff8b08`00000002 00000000`00000000 00000000`20707200 ffff8b08`8da02a60 : Wdf01000!FxPkgGeneral::Dispatch+0x1d673 [minkernel\wdf\framework\shared\irphandlers\general\fxpkggeneral.cpp @ 793] 
ffff9f83`1d7af210 fffff800`4e846d45     : ffff8b08`84f68300 00000000`00000000 ffff8b08`8f12b2a0 ffff8b08`8da02a60 : Wdf01000!FxDevice::DispatchWithLock+0x157 [minkernel\wdf\framework\shared\core\fxdevice.cpp @ 1447] 
ffff9f83`1d7af270 fffff800`ad3f1a6e     : ffff8b08`8f12b2a0 00000000`00000000 00000000`00000000 fffff800`4e846d87 : nt!IofCallDriver+0x55
ffff9f83`1d7af2b0 fffff800`4e846d45     : ffff8b08`88b13670 00000000`00000000 00000000`00000000 ffff9f83`1d7af4a9 : silabenm+0x1a6e
ffff9f83`1d7af2f0 fffff800`4ec0f30a     : 00000000`00000000 ffff8b08`88b13670 00000000`00000000 00000000`00040000 : nt!IofCallDriver+0x55
ffff9f83`1d7af330 fffff800`4ebf4f0c     : 00000000`00000000 00000000`00007fb2 00000000`00000000 ffff8b08`72313a60 : nt!IopCloseFile+0x17a
ffff9f83`1d7af3c0 fffff800`4ece1c25     : ffff8b08`7c352308 fffff800`4e822a79 ffff8b08`7c3522d0 00000000`00000000 : nt!ObCloseHandleTableEntry+0x24c
ffff9f83`1d7af500 fffff800`4ecc84d9     : ffff8b08`7c3522d0 ffff8b08`93acf500 ffffffff`ffffff01 ffff8b08`84f68758 : nt!ExSweepHandleTable+0xd5
ffff9f83`1d7af5b0 fffff800`4ecbb4bf     : ffffffff`ffffffff ffff8b08`84f68300 ffff9f83`1d7af600 ffff8b08`00000000 : nt!ObKillProcess+0x35
ffff9f83`1d7af5e0 fffff800`4ebe91d2     : ffff8b08`84f68300 ffffa10e`b210c6b0 ffff9f83`1d7af820 00000000`00000000 : nt!PspRundownSingleProcess+0x133
ffff9f83`1d7af660 fffff800`4ecf6518     : 00000000`00000000 00000000`00000001 00000000`00000000 00000000`00301000 : nt!PspExitThread+0x5f6
ffff9f83`1d7af760 fffff800`4e82a507     : 00000000`00000101 00000000`00000000 ffff8b08`85aee350 00000000`00000000 : nt!KiSchedulerApcTerminate+0x38
ffff9f83`1d7af7a0 fffff800`4e9e24e0     : 00000000`01aad4a0 ffff9f83`1d7af850 00000000`05c2fc78 ffff8b08`00000000 : nt!KiDeliverApc+0x487
ffff9f83`1d7af850 fffff800`4e9ef91f     : ffff8b08`93acf500 00000000`05c2fac8 00000000`00000000 00000000`01b20f28 : nt!KiInitiateUserApc+0x70
ffff9f83`1d7af990 00007ffc`d5dcf814     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nt!KiSystemServiceExit+0x9f
00000000`05c2fc28 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : 0x00007ffc`d5dcf814

SYMBOL_NAME:  silabser+5790

MODULE_NAME: silabser

IMAGE_NAME:  silabser.sys

STACK_COMMAND:  .thread ; .cxr ; kb

BUCKET_ID_FUNC_OFFSET:  5790

FAILURE_BUCKET_ID:  0x10D_5_silabser!unknown_function

OS_VERSION:  10.0.19041.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {79094898-163a-20f0-7919-7577c7df9c33}

Followup:     MachineOwner
---------
```

So it kind of looks like its related to some siliconlabs driver... I will try to find and uninstall it.

kittaakos, 2020-09-03:
> I never had any issues with the Java Arduino IDE

Good to know. Please note, the Java IDE does not use the CLI under the hood, the Pro IDE does.

> BLACKBOXWINLOGON: 1
>
> PROCESS_NAME:  arduino-cli.exe

Maybe we can narrow down the problem and see if you can reproduce the bluescreen issue when you compile and upload using the CLI only. Let me know if you need assistance.

ubidefeo, 2020-09-03:
hi @Blubbaa 
you wouldn't be the first to run into issues with boards using SiLab's CP2102 or CP2104 USB to TTL Serial adapters.
Silicon Labs has been falling behind when it comes to providing highly compatible drivers, and I have seen this a lot on Mac OS Catalina, a bit less on Windows.
Some users have complained in our forum about Arduino not supporting their ESP32 board, but there's nothing we can do about it.
We chose to design our boards around different USB interfaces that rarely require drivers to work, and sometimes need some tweaking on Linux (udev rules and such).
Hope you can fix your driver issue.
Sometimes it's good to remove the Windows' installed driver at the lowest level and download one from the manufacturer making sure you install it with admin rights

Blubbaa, 2020-09-03:
Hi @ubidefeo,
thank you for your thoughts! I also think the same about the Silabs CP2102 driver, I guess i will try to remove it if I experience any problems again, as the board I am currently using, does not need it.

ubidefeo, 2020-09-04:
@Blubbaa 
which USB controller does your board use?

Blubbaa, 2020-09-04:
The original board, which also caused the bluescreen, was an ESP32 Pico, which indeed needs the CP210x driver. I switched to using an Arduino with the CH340. Sorry I might have phrased that a little confusing.

ubidefeo, 2020-09-05:
@Blubbaa no worries :)