arduino/arduino-pro-ide #336: Save as > overwrite does not work.

kittaakos, 2020-09-04:
- Create a new sketch,
 - Save it as Foo,
 - Create a new sketch,
 - Save it as Foo,
 - IDE askes whether to overwrite (good ✅),
 - Hit yes (error ❌ ):

```
root ERROR Failed to execute command: Error: Request 'copy' failed
    at Proxy.<anonymous> (file:///Applications/Arduino%20Pro%20IDE.app/Contents/Resources/app/lib/bundle.js:60377:33)
    at SaveAsSketch.<anonymous> (file:///Applications/Arduino%20Pro%20IDE.app/Contents/Resources/app/lib/56.bundle.js:6956:65)
    at step (file:///Applications/Arduino%20Pro%20IDE.app/Contents/Resources/app/lib/56.bundle.js:6873:23)
    at Object.next (file:///Applications/Arduino%20Pro%20IDE.app/Contents/Resources/app/lib/56.bundle.js:6854:53)
    at fulfilled (file:///Applications/Arduino%20Pro%20IDE.app/Contents/Resources/app/lib/56.bundle.js:6845:58)
```

ubidefeo, 2020-09-05:
this works for me on Mac

kittaakos, 2020-09-05:
> this works for me on Mac

Thank you for trying. The trick is that you have to have a sketch named `Foo` and you have to save as `Foo`; it would rename itself. That was the bug.

ubidefeo, 2020-09-05:
@kittaakos 
this is exactly what I did
I _saved as_ my `LocationSender.ino` file over itself and it works
tried again just now

kittaakos, 2020-09-05:
> tried again just now

Which version of IDE2 are you using?

ubidefeo, 2020-09-06:
oops... @kittaakos testing the internal build
is it fixed there?

kittaakos, 2020-09-06:
> is it fixed there?

Yes. Thank you for the verification 👍

kittaakos, 2020-09-07:
Fixed. Will be available from the next nightly.

closed 2020-09-07 by kittaakos