arduino/arduino-pro-ide #337: [UX] The board select dialog is broken when the window size is small

kittaakos, 2020-09-04:
As a user, I can reduce the window size to 200X120 pixes but the board config dialog does not like it. It's a bug. See the screencast.

[image]

sbhklr, 2020-09-04:
@kittaakos For now I'd define a minimum size for the window. Something like 900x800?

kittaakos, 2020-09-07:
> For now I'd define a minimum size for the window. Something like 900x800?

It's done. From now on, the minimum window size is 900 x 800. (CC: @ubidefeo)

I leave the issue open until we have a proper fix.

paschlie, 2020-09-14:
Not sure if this comment is really necessary, but please note that sweeping this bug under the carpet by making the minimal window size large enough to mask the bug, shouldn't be considered acceptable; especially as the "solution" creates a bigger problem than it solves (i.e. the tool has now become unnecessarily effectively unusable for those using machines with smaller display dimensions typical of many laptops for example; where before it was merely a visual layout bug which didn't inhibit use of the tool).

kittaakos, 2021-02-25:
Originally posted by @LumitoLuma [here](https://github.com/arduino/arduino-pro-ide/issues/443#issue-816417125):

> **Describe the bug**
> I cannot see the full Arduino Pro IDE window on my Ubuntu 20.04 (GNOME) laptop. I also cannot maximize it.
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. Download Arduino Pro IDE 0.1.4 on Ubuntu 20.04
> 2. Extract
> 3. Open `arduino-pro-ide`
> 4. See error
> 
> **Expected behavior**
> Be able to fully resize the IDE window and maximize it.
> 
> **Screenshots**
> [image]
> 
> **Desktop (please complete the following information):**
> 
> * OS: Linux Mint (with GNOME desktop, based on Ubuntu 20.04.1 LTS)
> * Version: 20.1 Ulyssa
> 
> **Additional context**
> I also cannot vertically resize the window

kittaakos, 2021-02-25:
For ref: https://github.com/arduino/arduino-pro-ide/issues/351

kittaakos, 2021-03-01:
Originally reported by @ex-caliper [here](https://github.com/arduino/arduino-pro-ide/issues/448#issue-818543830):

> **Describe the bug**
> When you try to resize from either height or width, the IDE automatically increases the **height** beyond the
> bottom of the screen. Subsequently the bottom of the console cannot be seen.
> 
> **To Reproduce**
> Start the IDE as normal, try to reduce in anyway, height increases automatically beyond bottom of screen.
> Further attempts at recovering the height can result in the IDE minimising to the task bar and cannot be restored. It can be closed by right clicking the task bar icon, however, further attempts at restarting the IDE normally, automatically reduces it to the task bar rendering it unusable.
> 
> **Expected behavior**
> Resizing without issues mentioned.
> 
> **OS**: Windows 10 Pro
> **Version:** 18363
> 
> **Additional context**
> Further use of the IDE can only be achieved by deleting its folder and replacing with a fresh download.