arduino/arduino-pro-ide #338: The selected board + port should used for new sketches

kittaakos, 2020-09-04:
When I open a sketch and configure my board and port settings and I create a new sketch again, my previous board settings are not used for the new sketch; the app shows `no board selected`.

The sketches (browser windows) do not know anything about each other. We should put the selected board + port into the URL as a query, and when opening a new window, we should try to retrieve this info.

Erriez, 2020-09-10:
Related issue to my user experience [Examples menu missing Arduino Pro IDE #70](https://github.com/arduino/arduino-pro-ide/issues/70).

dalemcbeath, 2020-09-13:
I've been running into similar all morning and just found your notes/issue.   Sketches seem to have exclusive rights to a selected board/port. 

if I have a sketch open and have assigned a board/port, then want to open an existing sketch (new browser) and want to use that same board/port, it's not visible, not available, no ports found.  Hope that adds some additional info, if not, please delete.

Thanks, much appreciated by a nubbie.

ubidefeo, 2020-09-13:
@dalemcbeath 
>if I have a sketch open and have assigned a board/port, then want to open an existing sketch (new browser) and want to use that same board/port, it's not visible, not available, no ports found. Hope that adds some additional info, if not, please delete.

I cannot reproduce this.
I am able to select the board from the drop-down menu and it actually automatically switches to it if I had previously selected it

dalemcbeath, 2020-09-13:
When you say it automatically switches to it if you had previously selected it, do you still have the original browser window open?

Maybe it's because I'm trying to deal with an ESP32-Wrover module?  see #350  I haven't tried with normal Arduino boards, as my current project uses the ESP32 and is getting complicated to the point I decided to try the new IDE.  After I added the urls show in #350, The board showed after re-indexing.  When selected with it's port, the connected board does have an error triangle icon.  I can't find any way to see the error causing that - I've tried control clicking on the button, etc, looking for something obvious to a new user in the output....  Maybe whatever is causing the icon is also keeping it from being selected in the new project?   I've not been able to find any docs on what that icon means - I've also looked in the Theia docs.   There's a lot and I may have missed it.  Are there general IDE UI or existing project translation notes I'm missing?

Sorry for any trouble, and I really appreciate that someone out there is listening to my nubbie questions.

mahalo, 

dale

ubidefeo, 2020-09-13:
hi @dalemcbeath 
Sorry, I had not associated your issue to the other one :D

The auto-recognition of a board depends on its VID/PID (vendor ID and product ID) as a USB device.
When it comes to our boards (and some other manufacturers') we have a USB license which we pay for in order to have unique IDs for our boards. This allows us to develop a platform which accurately maps a device to a VID/PID pair.

Generic boards and modules will only be able to show up as a USB to Serial converter, but since all of them will contain the same pair we have no way to recognise which one is which, so a user has to manually select a board's type and its port.
The IDE tries to recognise this board and associate it automatically, which will only work with boards produced by manufacturers who go through the VID/PID registration process.

Hope this helps you understand where your issue comes from.
As you see this small UI bug has been assigned, so at some point you'll be able to just find your board pre-selected when opening a new sketch or switching to an already existing one.

Cheers.ubi

Erriez, 2020-09-13:
Just some feedback:

Checking the VID/PID registration process to find the serial port does not work for my custom hardware designs which uses  different serial interface chips. So this requires selecting the board / serial port manually. That's no problem as long as this setting is remembered when creating a new or opening an existing sketch.

>As you see this small UI bug has been assigned

A fix would be great.

> so at some point you'll be able to just find your board pre-selected when opening a new sketch or switching to an already existing one.

I think it needs some more clarification how I use the classic IDE:
* Start classic IDE
* Open a sketch, select board and serial port.
* From menu `File | New or Open` an existing sketch selects the same board and serial port.
* Upload from one of the two windows does not complain about the serial port is already in use.
* There is only one serial monitor window connected.

Now launch a second IDE (as a new instance from the start menu or command prompt).
* This way, you can select a different board and different serial port.
* Upload and serial monitor from this second IDE launch works independent from the first IDE launch.

With this approach, it is possible to develop a transmitter application in the first IDE and a receiver in the second IDE.

Do you think this is feasible for the pro IDE?

dalemcbeath, 2020-09-13:
thanks @ubidefeo, for the explanation, that helps. 

To make sure I understood what you meant, I wanted to just use a Mega board and a couple of projects.

- **open IDE**
    - one browser opens with old ESP project
    - open Example Blink
        - new browser opens _EXACTLY on top of existing one_, confusing, not standard UI for Mac
        - example loads
        - no board selected, pick Mega
        - SAVE
   - close old ESP browser, only the Mega browser open
- **QUIT**

- **open IDE**
    - one browser opens, the saved one, as expected, but **no board selected**.
        - try to pick board, only option is to Select Other Board and Port, pick that, no Mega.
        - _after about 2 minutes_ (while I'm taking notes) that changes and the ports appear, along with boards
        - I hit cancel.
        - hit no board selected dropdown again, shows Mega, pick it
        - save
- **QUIT**

- **open IDE**
    - opens with expected browser and has Mega selected
    - open 2nd browser example
        - CAN select same board.
        - SAVE
        - when save complete, **browser has lost board**, only option is to  Select Other Board and Port
        - as above, _after about 2 minutes_, the boards and ports appear, hit CANCEL and am able to pick Mega
- **QUIT** IDE with two browsers open, both with Mega board

- **open IDE**
    - ONE browser opens with Mega picked, other that was open when IDE closed is **missing**

Those 2 minute delays to find boards and ports seems odd. No offset to stacked browsers is not what I'm used to on OSx, but may be standard elsewhere.

ESP wise, it looks like I should wait another few mini-versions before attempting to migrate my ESP based project to the Pro IDE - there's just too many things missing (includes, ESP base libraries) for me to track down with my current knowledge level.  I'm very impressed with the new IDE, and it's already changed my coding style for the better.  Looking forward to using it on a larger project.

Again, thank you so much for your help in understand what was going on.  It's greatly appreciated.

dale

ubidefeo, 2020-10-21:
hi @Erriez and @dalemcbeath 

My apologies for the very late response (OMG it's been over a month).
My notifications got scrambled but thankfully I had kept the thread in my todo notes so I'm back here.

It's been a very busy month, but I'll go through the thread carefully later in the week and come back with a series of answers.

Thank you for your patience
ubi

ubidefeo, 2020-10-22:
@dalemcbeath 
I find this Mega drop-down behaviour very odd.
I'm having issue with my local configuration, but as far as I can tell when you have selected a board for a sketch it should remain.
We have yet to implement automatically selecting the last chosen board for a new sketch.
Also I'm with you on the main window opening right above the old one, it's not how it should work 👍

kittaakos, 2020-10-23:
> main window opening right above the old one, it's not how it should work

https://github.com/eclipse-theia/theia/issues/7431