arduino/arduino-pro-ide #339: "ctrl+T" changes the braces position I HATE that

Master811129, 2020-09-04:
after auto-fomating in Arduino Pro IDE. It changes the braces position (but legacy IDE doesn't):
[image]

ubidefeo, 2020-09-05:
hi @Master811129 
this is a setting in Clangd format that defaults to such behaviour

the Java IDE does not use ClangD for formatting.
I'm sure this can be tuned in the future, because everyone likes their style.
Arduino sketches are formatted according to a very specific set of rules in Astyle, so I guess that this behaviour will have to be tuned to reflect the one in the Java IDE.
Let's check wtih @per1234 and see what he has to say

kittaakos, 2020-09-07:
> after auto-fomating in Arduino Pro IDE. It changes the braces position (but legacy IDE doesn't):

Yes, this is a reasonable request. However, we have to figure out how users can/will customize it.
Let's investigate the `BraceWrappingFlags` options from [here](https://clang.llvm.org/docs/ClangFormatStyleOptions.html). [Here](https://zed0.co.uk/clang-format-configurator/) is an online `clangd` formatter with preview.