arduino/arduino-pro-ide #340: Unable to upload/compile for nano 33 ble

torenrl, 2020-09-04:
**Describe the bug**
Error while compiling for Arduino Nano 33 BLE

**To Reproduce**
Connect an Arduino 33 BLE and upload a sketch.

**Expected behavior**
Error message as follows:

Compiling...
{
  "sketchUri": "file://lipasto/kotidir03%24/ ... sketchname.ino",
  "fqbn": "arduino-beta:mbed:nano33ble",
  "optimizeForDebug": false,
  "port": "COM36"
}

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
Arduino Pro IDE 0.1.1

ubidefeo, 2020-09-05:
@torenrl 
you seem to have some weird character in your path.
can you try and move your sketch dir to another folder and test again?
thank you

torenrl, 2020-09-07:
@ubidefeo 

Removing the weird characters (i.e. moving the sketch to another location) seems to solve the problem. However, I have used the Pro IDE (previous version) for development of Arduino sketches earlier as well, with the same characters in the path. Uploading sketches were not a problem then with other Arduino boards, but I have not, until now, been able to upload to the nano 33 ble.

Thank you for the help. However, if there is a problem with the weird characters in the path, this would be nice to fix at some point. I am often working on non-local paths using non-english characters, and I guess I am not the only one doing this.

Again, thank you :)

kittaakos, 2020-09-07:
> if there is a problem with the weird characters in the path, this would be nice to fix at some point

Sure. It's a bug. Thanks for letting us know. @torenrl, can you please share the non-encoded FS path where the sketch was?
I am referring to `"file://lipasto/kotidir03%24/ ... sketchname.ino",`. What is the original URI without any modifications?

kittaakos, 2020-09-07:
> Error message as follows:
> 
> Compiling...
> {
> "sketchUri": "file://lipasto/kotidir03%24/ ... sketchname.ino",
> "fqbn": "arduino-beta:mbed:nano33ble",
> "optimizeForDebug": false,
> "port": "COM36"
> }

By the way, what is the error message? It seems like it got truncated. Can you please share the entire upload output? Thank you!

ubidefeo, 2020-09-07:
@torenrl 
thank you for testing this, as you might have read @kittaakos has picked this up already.

please provide all the info he needs in order to assist us fixing this bug 🙏

torenrl, 2020-09-14:
@ubidefeo and @kittaakos 
Thank you for looking into the issue. I have been unavailable lately, however, I will try to replicate the error and share the output within the next couple of days.

torenrl, 2020-09-14:
@kittaakos 
Strangely enough, I am not able to replicate the error, even though a just made a new sketch at the same location as I used while running into this problem. However, I do get an error (red x) box saying

> Could not start editor support. open \kotidir03$\ (username) \My Documents\Arduino\testsketch: The system cannot find the path specified.

kittaakos, 2020-09-14:
> he symbols in the path are exactly as written here, when looking from the explorer and also from the command line. It is on a remote disk, starting with \\lipasto\kotidir... instead of f.ex. D:\ or C:\ etc.

Thanks for the update. I am going to leave this issue open. I am sure there is something wrong with the URIs: it starts with `"sketchUri": "file://l` and it should be `"sketchUri": "file:///l` (Note, the three slashes instead of two.)

ubidefeo, 2021-02-17:
@kittaakos we're closing this one, as we haven't had any report of upload failing for BLE on Windows in the last few months

closed 2021-02-17 by ubidefeo