arduino/arduino-pro-ide #341: https://downloads.arduino.cc/arduino-pro-ide/nightly/CHANGELOG.txt - 0.1.2?

jwestmoreland, 2020-09-04:
Hello,

From the current changelog:

There were no changes since version [0.1.2](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.2).

So, obvious question is - when will 0.1.2 be posted?

Thanks,
John W.

ubidefeo, 2020-09-05:
@jwestmoreland 
we're doing some cleanup and addressing bugs, don't have an ETA for 0.1.2 yet.
Nightly builds won't change the release number, releases will still be manually tagged.
In the meantime Martino and I have been looking at ThreadDebug and it works in the CLI, so investigation in why it doesn't trigger in the IDE is ongoing.
Just keep in mind that the debug UI and engine in the new IDE won't change much, but we're still looking at how to replace it entirely with something else.
Trust me when I say you'll be the first to know :D

jwestmoreland, 2020-09-05:
Hello @ubidefeo,

I see the changelog has [changed.](https://downloads.arduino.cc/arduino-pro-ide/nightly/CHANGELOG.txt)

I've been looking over [here:](https://github.com/arduino/ArduinoCore-mbed/issues/50) - and I haven't been able to get the driver not to recognize the Portenta - I've even updated to the latest commit/plus .elf file - ran some tests, and reverted.

The Portenta did enumerate to new COM ports - that's about it.  Instead of seeing two COM4 ports like discussed [here:](https://github.com/arduino/ArduinoCore-mbed/issues/46) - I'm seeing COM17 and COM19 there now, respectively.

Any hints regarding what the new debugger will be?  Will it still be integrated with the new IDE?  Just wondering.

Ha!  Thanks for that - looking forward to what's coming down the pike.

Thanks,
John W.

ubidefeo, 2020-09-05:
@jwestmoreland 
the new core (master branch) implements an alias for the debug port so the enumeration changes are justified.
the new debugger will be part of the IDE, that's one of our main proposition :)
we're investigating a couple of VS Code plugins because they are supported by Theia and moving forward we will try and include such plugins when it comes to new features, but some UI issues might arise that need to be addressed.

I know I can count on you to have brutal feedback on the Debugger :D

jwestmoreland, 2020-09-05:
Hello @ubidefeo ,

> the new core (master branch) implements an alias for the debug port so the enumeration changes are justified.
OK - that's good.  (There could be an issue with bootloader mode with the latest commit over [here:](https://github.com/arduino/ArduinoCore-mbed/issues/50)

> the new debugger will be part of the IDE, that's one of our main proposition :)
> we're investigating a couple of VS Code plugins because they are supported by Theia and moving forward we will try and [...]

Sounds good.

> I know I can count on you to have brutal feedback on the Debugger :D

Ha!  Ok - I'll try!

Thanks,
John W.

kittaakos, 2020-09-07:
@jwestmoreland thank you for the heads-up about the misleading 0.1.2 changelog. We had to test the entire CI pipeline, so I drafted a 0.1.2 dummy tag and then removed it.

The current changelog correctly shows the diff between 0.1.1 and the current `HEAD`.

I am going to close this issue. If you think there is anything we can and have to do regarding the changelog, please let me know and I will reopen the issue. Thank you!

closed 2020-09-07 by kittaakos