arduino/arduino-pro-ide #344: Not able to load workspace

DeeEmm, 2020-09-08:
What's happened to workspaces?

**Describe the bug**
Not able to load (theia) workspace
Workspace file tools missing from file menu (load workspace / save workspace)
Existing workspaces are not loading correctly - Root folder of project / workspace is incorrect: Project opens at arduino folder and not root folder of stored project. This makes it impossible to manage project files & documents stored above the arduino folder level (for example github documents)

 - OS: OSX 10.15.6
 - Version 0.1.1

**Additional context**
Not sure if this is simply an incompatibility between previously saved workspace but I am unable to resolve the issue using the controls within the IDE as previously accessible workspace tools are no longer present.

I suspect that this may be an intentional change? To simplify the arduino experience? To standardise arduino folder architecture? If this is the case I think that it would be good to retain the previous behaviour as an option so that those who want to use the IDE to manage more complex projects are able to do so.

If this is the case I propose that the workspace controls are able to be reinstated selectively using a configuration setting. 

/DM

kittaakos, 2020-09-08:
> I suspect that this may be an intentional change? To simplify the arduino experience? To standardise arduino folder architecture? If this is the case I think that it would be good to retain the [...]

Yes, we have intentionally removed it to align the UX with the Java-based Arduino IDE. See #317.

I am going to convert this to a feature request.

DeeEmm, 2020-09-08:
Thanks @kittaakos.