arduino/arduino-pro-ide #346: Auto Syntax / auto complete / Contex relevant help

Docara, 2020-09-08:
Hi

As mentioned elsewhere I am new to the Arduino C++ programming world, A thing which I find extremely unhelpful is the lack of proper help support for commands whilst typing. If I do not know the proper syntax of a command, hitting a search engine etc and researhing  etc etc.

What I would like is an IDE to help me whilst coding - be interactive with me - hold my hand so I don't make mistakes and help me learn. The standard IDE ( as the majority of other IDE's are) are particularly unhelpful. They remind me of old text based word processors before Word and graphical User Interfaces came along.

ubidefeo, 2020-09-09:
hi @Docara 

I understand where you're coming from, we have all been there at one point.
There is a lot of research in the field of interactive programming environments (Bret Victor comes to mind, and results of his work can be seen in places such as Swift Playgrounds), and I'm sure one day every code editor will offer such capabilities, but this is not in our near future.

We do offer code-assist and completion as well as inline labels that tell you a bit more about the method you're about to invoke, but this is still not entirely human friendly by the nature of the underlying language.
Have you used code assist in our latest nightly build? Try type 
What we do at Arduino is help the beginner with examples, documentation and tutorials.
That is to me what can help you overcome the initial limitations of not knowing a language.
Also our way of defining objects and functions names (API) has been simplified for beginners to feel more comfortable with.

What I always tell my students is that if you are interested in and intrigued by something you'll get to understand it, and at some point you'll love computers enough to want to talk to them... in their own language :)

Thank you for offering your suggestion, I hope that

ubidefeo, 2021-02-17:
Too much out of scope for this project

closed 2021-02-17 by ubidefeo

per1234, 2021-02-18:
Related feature request: https://github.com/arduino/arduino-pro-ide/issues/77