arduino/arduino-pro-ide #350: errors loading additional url in board manager

dalemcbeath, 2020-09-12:
**Describe the bug**
re: issue 61 found searching for info on how to add ESP32 board to Pro IDE, added URL as shown, which produces errors.

**To Reproduce**
board_manager:
  additional_urls: [https://dl.espressif.com/dl/package_esp32_index.jsonc]

**Expected behavior**
expected board I am using in regular IDE to work in Pro version

**Desktop (please complete the following information):**
  Mac 10.15.6

**Additional context**
board is an Espresiff ESP32-WROVER-B, ,working fine from normal IDE

**ERRORS** 
[INFO] Checking signature [index: /Users/CritterLab/Library/Arduino15/package_index.json, signaturefile: /Users/CritterLab/Library/Arduino15/package_index.json.sig, trusted: true]
[INFO] Checking if CLI is Bundled into the IDE
[INFO] Loading hardware from: /Users/CritterLab/Library/Arduino15/packages
[INFO] Loading package Heltec-esp32 from: /Users/CritterLab/Library/Arduino15/packages/Heltec-esp32/hardware
[INFO] Loaded platform [platform: {ID:"Heltec-esp32:esp32",Installed:"0.0.5",Latest:"0.0.5"}]
[INFO] Checking existence of 'tools' path: /Users/CritterLab/Library/Arduino15/packages/Heltec-esp32/tools
[INFO] Loading tools from dir: /Users/CritterLab/Library/Arduino15/packages/Heltec-esp32/tools
[INFO] Loaded tool [tool: Version: 2.6.1]
[INFO] Loaded tool [tool: Version: 0.2.3]
[INFO] Loaded tool [tool: Version: 1.22.0-80-g6c4433a-5.2.0]
[INFO] Loading package arduino from: /Users/CritterLab/Library/Arduino15/packages/arduino/hardware
[INFO] Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.3",Latest:"1.8.3",Name:"Arduino AVR Boards"}]
[INFO] Loaded platform [platform: {ID:"arduino:mbed",Installed:"1.1.4",Latest:"1.1.6",Name:"Arduino nRF528x Boards (Mbed OS)"}]
[INFO] Loaded platform [platform: {ID:"arduino:megaavr",Installed:"1.8.6",Latest:"1.8.6",Name:"Arduino megaAVR Boards"}]
[INFO] Loaded platform [platform: {ID:"arduino:sam",Installed:"1.6.12",Latest:"1.6.12",Name:"Arduino SAM Boards (32-bits ARM Cortex-M3)"}]
[INFO] Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.6",Latest:"1.8.8",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
[INFO] Loaded platform [platform: {ID:"arduino:samd_beta",Installed:"1.6.25",Latest:"1.6.25",Name:"Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)"}]
[INFO] Checking existence of 'tools' path: /Users/CritterLab/Library/Arduino15/packages/arduino/tools
[INFO] Loading tools from dir: /Users/CritterLab/Library/Arduino15/packages/arduino/tools
[INFO] Loaded tool [tool: Version: 4.5.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Loaded tool [tool: Version: 1.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
[INFO] Loaded tool [tool: Version: 1.2.1 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 1.3.0 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 4.8.3-2014q1 Systems: [OS: arm-linux-gnueabihf, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loaded tool [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loaded tool [tool: Version: 7.3.0-atmel3.6.1-arduino5 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin14, OS: i686-mingw32, OS: i686-linux-gnu, OS: x86_64-linux-gnu]]
[INFO] {"level":"info","msg":"Loaded tool","time":"2020-09-11T15:29:48-10:00","tool":{"version":"7.3.0-atmel3.6.1-arduino7","systems":[{"os":"arm-linux-gnueabihf","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-arm-linux-gnueabihf.tar.bz2","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-arm-linux-gnueabihf.tar.bz2","Checksum":"SHA-256:3903553d035da59e33cff9941b857c3cb379cb0638105dfdf69c97f0acc8e7b5","Size":34683056,"CachePath":"packages"}},{"os":"aarch64-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-aarch64-pc-linux-gnu.tar.bz2","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-aarch64-pc-linux-gnu.tar.bz2","Checksum":"SHA-256:03d322b9df6da17289e9e7c6233c34a8535d9c645c19efc772ba19e56914f339","Size":38045723,"CachePath":"packages"}},{"os":"x86_64-apple-darwin14","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-apple-darwin14.tar.bz2","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-apple-darwin14.tar.bz2","Checksum":"SHA-256:f6ed2346953fcf88df223469088633eb86de997fa27ece117fd1ef170d69c1f8","Size":36684546,"CachePath":"packages"}},{"os":"i686-mingw32","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-w64-mingw32.zip","Checksum":"SHA-256:a54f64755fff4cb792a1495e5defdd789902a2a3503982e81b898299cf39800e","Size":52519412,"CachePath":"packages"}},{"os":"i686-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-pc-linux-gnu.tar.bz2","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-i686-pc-linux-gnu.tar.bz2","Checksum":"SHA-256:954bbffb33545bcdcd473af993da2980bf32e8461ff55a18e0eebc7b2ef69a4c","Size":37176991,"CachePath":"packages"}},{"os":"x86_64-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-pc-linux-gnu.tar.bz2","ArchiveFileName":"avr-gcc-7.3.0-atmel3.6.1-arduino7-x86_64-pc-linux-gnu.tar.bz2","Checksum":"SHA-256:bd8c37f6952a2130ac9ee32c53f6a660feb79bee8353c8e289eb60fdcefed91e","Size":37630618,"CachePath":"pack
[INFO] ages"}}]}}
[INFO] Loaded tool [tool: Version: 6.3.0-arduino17 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: x86_64-apple-darwin12, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 1.6.1-arduino Systems: [OS: arm-linux-gnueabihf, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
[INFO] Loaded tool [tool: Version: 1.7.0-arduino3 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 1.9.1-arduino1 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 0.9.0-arduino2 Systems: [OS: i386-apple-darwin11, OS: arm-linux-gnueabihf, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] Loaded tool [tool: Version: 0.10.0-arduino13 Systems: [OS: aarch64-linux-gnu, OS: arm-linux-gnueabihf, OS: i686-linux-gnu, OS: i686-mingw32, OS: i386-apple-darwin11, OS: x86_64-linux-gnu]]
[INFO] Loaded tool [tool: Version: 0.10.0-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
[INFO] {"level":"info","msg":"Loaded tool","time":"2020-09-11T15:29:48-10:00","tool":{"version":"0.10.0-arduino9","systems":[{"os":"aarch64-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-aarch64-linux-gnu.tar.bz2","ArchiveFileName":"openocd-0.10.0-arduino9-static-aarch64-linux-gnu.tar.bz2","Checksum":"SHA-256:b814b16b52cef21eacf456cc7a89d7b5d4cb1385bfb8fb82963b7d8151824d93","Size":1714657,"CachePath":"packages"}},{"os":"arm-linux-gnueabihf","Resource":{"URL":"http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-arm-linux-gnueabihf.tar.bz2","ArchiveFileName":"openocd-0.10.0-arduino9-static-arm-linux-gnueabihf.tar.bz2","Checksum":"SHA-256:f0443e771f5e3a779949498d3c9bfffd1dd27cdf0ad7136a2db5e80447a7175a","Size":1778177,"CachePath":"packages"}},{"os":"i686-linux-gnu","Resource":{"URL":"http://downloads.arduino.cc/tools/openocd-0.10.0-arduino9-static-i686-ubuntu12.04-linux-gnu.tar.bz2","ArchiveFileName":"openocd-0.10.0-arduino9-static-i686-ubuntu12.04-linux-gnu.tar.bz2","Checksum":"SHA-256:a22872918df899cb808f9286141d42438ae5611156c143cfb692069f52a2bddd","Size":1 [... cut ...]
[INFO] Adding libraries dir [dir: /Users/CritterLab/Library/Arduino15/packages/Heltec-esp32/hardware/esp32/0.0.5/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/CritterLab/Library/Arduino15/packages/arduino-beta/hardware/mbed/1.2.2/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/CritterLab/Library/Arduino15/packages/atmel-avr-xminis/hardware/avr/0.6.0/libraries, location: platform]
[INFO] Adding libraries dir [dir: /Users/CritterLab/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries, location: platform]**

ubidefeo, 2020-09-12:
the easiest way to make this work is to add the list of additional urls in this format
```
additional_urls:
    - "https://dl.espressif.com/dl/package_esp32_index.json"
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
    - https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
```

closed 2020-09-12 by ubidefeo

dalemcbeath, 2020-09-13:
Thanks for the insights!

For anyone else finding this, here's what allowed me to load ESP 32 board info:

From Arduino Pro IDE main menu, pick Open CLI Configuration, and add additional URL's - this is the only format I could get to work.

board_manager:
```
  additional_urls: [
     https://dl.espressif.com/dl/package_esp32_index.json,
     http://arduino.esp8266.com/stable/package_esp8266com_index.json,
     https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
  ]
```

Be careful if copying from an email, text file, or from another editor, as you may end up with unrecognized characters
(such as special reverse double quotes), because the IDE will hang on restart.  To edit the file manually, use a terminal find your user/.arduinoProIDE folder, and edit the arduino-cli.yaml file to remove type-o or change values.  

Once done, use update commands in this order:

    arduino-cli lib update-index
    arduino-cli core update-index

and watch for errors, correct and repeat as needed.  When updates run smoothly, restart the IDE.