arduino/arduino-pro-ide #351: arduino-pro-ide_nightly-20200912_macOS_64bit windows resized larger than screen

paschlie, 2020-09-12:
Open a file and then attempt to resize its windo, and discover it can't be reasonably arbitrarily resized; where in my case the window becomes taller than available screen dimensions, and then can't be reduced, and with it's title bar becoming forced to the top of the screen, and its bottom off screen.

ubidefeo, 2020-09-12:
hi @paschlie 
please make sure you fill in the bug report correctly according to our template (which is available to you when you create a bug issue

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

paschlie, 2020-09-12:
The problem appears to have been caused by:
Changes since version [0.1.1](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.1):
 - arduino/arduino-pro-ide#337: Use bigger min window [c3eb3e4]
The Macbook-Pro I'm using has a screen resolution of: 1280 x 800 (including the menu bar)

ubidefeo, 2020-09-13:
@paschlie 
I remember @sbhklr reporting it and Akos fixing, so I thought this was another one.
Have you tried with the nightly build?
Again, this is why filling the whole issue is necessary

paschlie, 2020-09-13:
I'm guessing the "fix" was merely to increase the min-window size, but was made too large, so that now the window can't be made smaller than the screen on some displays.  (As there's really only one way to attempt to resize a window displayed on screen, there didn't seem to be much point to explain how to do it in more detail, sorry.)  Please see the top of the resulting displayed window pegged afterward to the top of the screen, the the bottom of the window falling below the bottom of screen out of view:

[image]

sbhklr, 2020-09-13:
@paschlie Thanks for reporting. We'll look into it next week.

kittaakos, 2021-02-11:
Duplicate of https://github.com/arduino/arduino-pro-ide/issues/337. Closing.

closed 2021-02-11 by kittaakos