arduino/arduino-pro-ide #354: Please get rid of the "outline view", as code should be documented in-line;

paschlie, 2020-09-13:
not in/via some appendage which seemingly needlessly takes up screen real estate, presumably longer-term meant to enable reference what should be documented within the code itself, I'm guessing?

(Being able to easily open included header files, etc. for example, is likely a much more useful feature for most users; as would to be able to easily inspect/edit arbitrary library source files from within the IDE; I believe.)

sstaub, 2020-09-14:
Do it like VSCode please

kittaakos, 2020-09-14:
Thanks for letting us know.

For the record: the default, classic layout does not include the Outline view; if you want to gain an additional 50 px, you can right-click on it and select _Close_.

[image]

paschlie, 2020-09-14:
> Thanks for letting us know.
> 
> For the record: the default, classic layout does not include the Outline view; if you want to gain an additional 50 px, you can right-click on it and select _Close_.

Thank you, however no such thing appears on macos 10.13.6 right-click (presumably a ctrl-click) to remove the right-hand control icon column, and corresponding outline control icon positioned there,  the only thing any type of click on the outline icon does is to toggle the display of the outline pane, which doesn't itself appear to work yet.

Maybe this is should really be a personal choice, users should be able to decide whether to place icons to enable certain features on the right or left hand columns, and such that if none are placed in one or the other, the space reverts for use by the editor/console sub-windows.  But if too complicated, then maybe all control icons should be placed in the left hand icon column, as someone else noted VSCode does; as having both if non-configurable, arguably just takes up space and adds little if any utility.

(As an aside, the minimum window sizes have already been made too large to be properly displayed on smaller screens; saving strictly unnecessary use of window space can only help the situation.)

kittaakos, 2020-09-15:
> however no such thing appears on macos

Strange, it works for me:

[image]

paschlie, 2020-09-15:
Please notice the "outline view" icon is always present without any way to disable it with any key-click combination I can find:
[image]

And when "outline view" is displayed, it does not contain anything useful on my installation, and never noticed it ever doing so?:
[image]

kittaakos, 2020-09-15:
> And when "outline view" is displayed, it does not contain anything useful on my installation, and never noticed it ever doing so?:

You have to pick a board to start the language server and have the _Outline_ view populated. (See the specification for `'textDocument/documentSymbol'` [here](https://microsoft.github.io/language-server-protocol/specification.html#textDocument_documentSymbol) if you're interested.)

paschlie, 2020-09-15:
Selecting a board doesn't seem to make any difference, but if significant here are the messages present after selecting the board (and not sure why it should make a difference, but why can't the "outline view" icon and corresponding right-hand icon column be dismissed, does it somehow rely on "outline-view" working properly?):
[image]

paschlie, 2020-09-15:
[2020-09-15_log.log](https://github.com/arduino/arduino-pro-ide/files/5225462/2020-09-15_log.log)

kittaakos, 2020-09-16:
> so its presence won't unnecessarily use screen real estate on the right side of the window.

> Do it like VSCode please

Thank you for your feedback; I moved it to the left side.

Classic mode:
[image]

Advanced mode:
[image]

The change will be available with one of the nightly builds this week or next week.

paschlie, 2020-09-16:
Has this change been published in the nightly build? (I didn’t notice any change log entry?)

kittaakos, 2020-09-16:
> Has this change been published in the nightly build? (I didn’t notice any change log entry?)
> […](#)

No, not yet. 👇 

> The change will be available with one of the nightly builds this week or next week.

paschlie, 2020-09-16:
> … <x-msg://58/#>

Thanks, I’m assuming from your posting that if “outline view” is toggled on, its icon remains in the left hand icon doc, from where its view pane may be expanded or contracted with a click on its icon as may be desired?

closed 2021-02-11 by kittaakos