arduino/arduino-pro-ide #355: Please Recognize Programmers In ./1.2.2/programmers.txt

jwestmoreland, 2020-09-15:
Hello All,

It appears the 0.1.2 IDE does not recognized programmers.txt in the ./1.2.2/programmers.txt file.

So, in the current IDE - how is that being handled?

There's a related PR [here FYI.](https://github.com/arduino/ArduinoCore-mbed/pull/28)

Thanks,
John W.

kittaakos, 2020-10-12:
Dev notes:

IDE2 should handle when installing cores manually at the `${directories.user} location`:
```
git clone --depth 1 https://github.com/arduino/ArduinoCore-mbed hardware/arduino-beta-development/mbed
git clone --depth 1 https://github.com/arduino/ArduinoCore-API hardware/arduino-beta-development/ArduinoCore-API
```

Then IDE2 will incorrectly mark two separate boards as duplicate.

`arduino:mbed@1.3.0`:
```json
{
  "packages": [
    {
      "name": "arduino",
      "maintainer": "Arduino",
      "websiteURL": "http://www.arduino.cc/",
      "email": "packages@arduino.cc",
      "help": {
        "online": "http://www.arduino.cc/en/Reference/HomePage"
      },
      "platforms": [
        {
          "name": "Arduino mbed-enabled Boards",
          "architecture": "mbed",
          "version": "1.3.0",
          "category": "Contributed",
          "url": "http://downloads.arduino.cc/cores/ArduinoCore-mbed-1.3.0.tar.bz2",
          "archiveFileName": "ArduinoCore-mbed-1.3.0.tar.bz2",
          "checksum": "SHA-256:f6617b34b53af37dbc2e6805097cf961f2032dbfc3750a003ba1902d7469258b",
          "size": "93357731",
          "help": {
            "online": "https://github.com/arduino/ArduinoCore-mbed/issues"
          },
          "boards": [
            {
              "name": "Arduino Portenta H7"
            },
            {
              "name": "Arduino Nano 33 BLE"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "openocd",
              "version": "0.10.0-arduino13"
            },
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "7-2017q4"
            },
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.9.1-arduino2"
            },
            {
              "packager": "arduino",
              "version": "0.9.0-arduino2",
              "name": "dfu-util"
            }
          ]
        }
      ]
    }
  ]
}
```

and

`arduino-beta:mbed@1.3.0`:
```json

  "packages": [
    {
      "name": "arduino-beta",
      "maintainer": "Arduino",
      "websiteURL": "http://www.arduino.cc/",
      "email": "packages@arduino.cc",
      "help": {
        "online": "http://www.arduino.cc/en/Reference/HomePage"
      },
      "platforms": [
        {
          "name": "Arduino mbed-enabled Boards",
          "architecture": "mbed",
          "version": "1.3.0",
          "category": "Contributed",
          "url": "http://downloads.arduino.cc/cores/ArduinoCore-mbed-1.3.0.tar.bz2",
          "archiveFileName": "ArduinoCore-mbed-1.3.0.tar.bz2",
          "checksum": "SHA-256:f6617b34b53af37dbc2e6805097cf961f2032dbfc3750a003ba1902d7469258b",
          "size": "93357731",
          "help": {
            "online": "https://github.com/arduino/ArduinoCore-mbed/issues"
          },
          "boards": [
            {
              "name": "Arduino Portenta H7"
            },
            {
              "name": "Arduino Nano 33 BLE"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "openocd",
              "version": "0.10.0-arduino13"
            },
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "7-2017q4"
            },
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.9.1-arduino2"
            },
            {
              "packager": "arduino",
              "version": "0.9.0-arduino2",
              "name": "dfu-util"
            }
          ]
        }
      ]
    }
  ]
}
```

Error in IDE2:
```
root ERROR Warning: Encountered two children with the same key, `Arduino Nano 33 BLE-Arduino mbed-enabled Boards`. Keys should be unique so that components maintain their identity across updates. Non-unique keys may cause children to be duplicated and/or omitted — the behavior is unsupported and could change in a future version.
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in div (created by BoardsConfig)
    in BoardsConfig
    in div
```

------

[package_index.json.txt](https://github.com/arduino/arduino-pro-ide/files/5367540/package_index.json.txt)

kittaakos, 2020-10-13:
> Dev notes:

I have fixed the bug. It will be available with one of the releases. From now on, we will show the `ID` of the core instead of its `name` when there are duplicate board names.

Before:
[image]

After:
[image]

kittaakos, 2020-12-07:
This request is a little bit problematic, so I hope to get some instructions from the CLI team on tackling this issue. (@cmaglie, @ubidefeo)

Request:
According to the initial feature request of this issue, users would like to see the boards from the manually installed platforms in the boards dialog. 

Current behavior:
To list all boards, including those that are not yet installed, the boards dialog uses the `core search` functionality under the hood. `core search` does not include boards from manually installed platforms.

Proposed solution:
We have to run `board listall` to include the boards from manually installed platforms, and we have to run `core search` for the current behavior and "just" concatenate the two results sets.

Problem:
The name of the board can change after installing the corresponding platform.

Steps to reproduce the issue:
 - Wipe your `~/Library/Arduino15` content.
 - Install the AVR core manually: `git clone --depth 1 https://github.com/arduino/ArduinoCore-avr.git ~/Documents/Arduino/hardware/arduino/avr`.
 - Run `board listall` and `core search` and try to concatenate the resultsets.

`./arduino-cli board listall --format json`:
```json
{
  "boards": [
    // [...]
    {
      "name": "Arduino Mega or Mega 2560",
      "FQBN": "arduino:avr:mega"
    },
    {
      "name": "Arduino Mega ADK",
      "FQBN": "arduino:avr:megaADK"
    },
    // [...]
  ]
}
```

`./arduino-cli core search "Arduino Mega" --format json`:
```json
[
  {
    "ID": "arduino:avr",
    "Latest": "1.8.3",
    "Name": "Arduino AVR Boards",
    "Maintainer": "Arduino",
    "Website": "http://www.arduino.cc/",
    "Email": "packages@arduino.cc",
    "Boards": [
      // [...]
      {
        "name": "Arduino Mega"
      },
      {
        "name": "Arduino MegaADK"
      },
      // [...]
    ]
  }
]
```

I could not find a way to associate one unique key for the same concept: `arduino:avr:mega` from the two result sets, so the board would show up twice in the dialog. Any ideas on how this should work with the currently available set of CLI commands? Thanks!

per1234, 2021-02-19:
Hi @jwestmoreland. Thanks for taking the time to request this important feature.

I see that over the course of the investigation the problem was found to be the lack of full support for boards platforms installed in the sketchbook folder. We already have a prior issue report about this at https://github.com/arduino/arduino-pro-ide/issues/172.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of that one.
If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-02-24 by kittaakos