arduino/arduino-pro-ide #356: The language server does not start on macOS

kittaakos, 2020-09-15:
Originally reported by @paschlie [here](https://github.com/arduino/arduino-pro-ide/issues/354#issuecomment-692732812):

> Selecting a board doesn't seem to make any difference, but if significant here are the messages present after selecting the board (and not sure why it should make a difference, but why can't the "outline view" icon and corresponding right-hand icon column be dismissed, does it somehow rely on "outline-view" working properly? And the bottom of my window is still cut off as it's minimum dimensions are larger than the screen):
> [image]

kittaakos, 2020-09-15:
@paschlie, can you please check the log files and share the content with us? The files are under `~/Library/Logs/Arduino Pro IDE/YYYY-MM-DD_log.log`. Thanks!

kittaakos, 2020-09-15:
Relevant parts from [the log](https://github.com/arduino/arduino-pro-ide/issues/354#issuecomment-692739091):
```
2020-09-15 09:26:55 daemon INFO Checking signature [index: /Users/paul/Library/Arduino15/package_index.json, signaturefile: /Users/paul/Library/Arduino15/package_index.json.sig, trusted: true]
2020-09-15 09:26:55 daemon INFO Checking signature [error: opening signature file: open /Users/paul/Library/Arduino15/package_seeeduino_boards_index.json.sig: no such file or directory, index: /Users/paul/Library/Arduino15/package_seeeduino_boards_index.json, signaturefile: /Users/paul/Library/Arduino15/package_seeeduino_boards_index.json.sig]
2020-09-15 09:26:55 daemon INFO Checking signature [error: opening signature file: open /Users/paul/Library/Arduino15/package_adafruit_index.json.sig: no such file or directory, index: /Users/paul/Library/Arduino15/package_adafruit_index.json, signaturefile: /Users/paul/Library/Arduino15/package_adafruit_index.json.sig]
2020-09-15 09:26:55 daemon INFO Checking if CLI is Bundled into the IDE
```

```
2020-09-15 09:27:02 daemon INFO Checking signature [error: opening signature file: open /var/folders/_k/1xs6gysd6glfxlhnjfx8s5cw0000gn/T/121662386.sig: no such file or directory, index: /var/folders/_k/1xs6gysd6glfxlhnjfx8s5cw0000gn/T/121662386, signaturefile: /var/folders/_k/1xs6gysd6glfxlhnjfx8s5cw0000gn/T/121662386.sig]
```

```
2020-09-15 09:27:07 root INFO Index has been updated.
2020-09-15 09:27:07 discovery INFO Updated discovered boards and available ports. - Discovered boards: [] and available ports: [{"protocol":"serial","address":"/dev/cu.PaulsRugby-BluetoothSer"},{"protocol":"serial","address":"/dev/cu.SGH-A847-SerialServer"},{"protocol":"serial","address":"/dev/cu.Bluetooth-Incoming-Port"}]
2020-09-15 09:27:09 root INFO Attached boards and available ports changed: 
2020-09-15 09:27:11 root ERROR Arduino: 2020/09/15 09:27:10 Command error: /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli exit status 1
2020/09/15 09:27:10 ------------------------------BEGIN STDERR
 /private/var/folders/_k/1xs6gysd6glfxlhnjfx8s5cw0000gn/T/ino2cpp-794929240/Feather_LoRa_Gateway/Feather_LoRa_Gateway.ino:29:10: fatal error: WiFi_AP.h: No such file or directory
   29 | #include "WiFi_AP.h"        // WiFi AP cridentials
      |          ^~~~~~~~~~~
compilation terminated.
Error during build: exit status 1
 ------------------------------END STDERR
```

```
2020-09-15 09:27:35 root ERROR Arduino: 2020/09/15 09:27:35 Command error: /Applications/Arduino Pro IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli exit status 1

2020-09-15 09:27:35 root ERROR Arduino: 2020/09/15 09:27:35 ------------------------------BEGIN STDERR
 Error during build: platform not installed
 ------------------------------END STDERR

2020-09-15 09:30:33 root ERROR Arduino: 2020/09/15 09:30:33 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:32:01 root ERROR Arduino: 2020/09/15 09:32:01 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:32:53 root ERROR Arduino: 2020/09/15 09:32:53 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:33:11 root ERROR Arduino: 2020/09/15 09:33:11 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:33:14 root ERROR Arduino: 2020/09/15 09:33:14 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:33:54 root ERROR Arduino: 2020/09/15 09:33:54 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:41:06 root INFO >>> Disposing boards service...
2020-09-15 09:41:06 root INFO <<< Disposed boards service.
2020-09-15 09:41:06 monitor-service INFO >>> Disposing monitor service...
2020-09-15 09:41:06 monitor-service INFO <<< Disposed monitor service.
2020-09-15 09:41:06 root INFO >>> Disposing library service...
2020-09-15 09:41:06 root INFO <<< Disposed library service.
2020-09-15 09:41:06 root INFO >>> Disposing boards service...
2020-09-15 09:41:06 root INFO <<< Disposed boards service.
2020-09-15 09:43:22 root ERROR Arduino: 2020/09/15 09:43:22 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:47:38 root ERROR Arduino: 2020/09/15 09:47:38 jsonrpc2 handler: notification "$/cancelRequest" handling error: jsonrpc2: connection is closed

2020-09-15 09:47:56 root INFO The CLI daemon process has started.
```

paschlie, 2020-09-15:
If helpful, I had two projects open:

- one which I was working on which references “WiFi_AP.h” which although the IDE has always warned about, everything has complied fine (so although it claims not to be found, it obviously was, and was in fact present in the same directory as the projects source .ino file).

- followed by the second being the basic new project ino file opened to duplicate your example.

Neither of which appeared to produce the expected “outline view” results.

kittaakos, 2020-09-15:
> If helpful,

Yes, it is. Thank you!

ubidefeo, 2021-02-17:
not an issue anymore
please try `0.1.4` in order to verify

closed 2021-02-17 by ubidefeo