arduino/arduino-pro-ide #361: Cannot upload sketch

czghzc, 2020-09-23:
Cannot upload sketch

Steps to reproduce the behavior:
1. Go to 'Examples'
2. Click on '01.Basics' --> 'Blink'
3. Select Arduino Nano board  and Com Number
4. Click on 'Upload'

Desktop
 - OS: [Win7-64bit]
 - Version: [Arduino Pro IDE 0.1.2]

Uploading...
{
  "sketchUri": "file:///c%3A/Users/admin/AppData/Local/Temp/.arduinoProIDE-unsaved2020823-6020-1lngmmb.n49cj/Blink/Blink.ino",
  "fqbn": "arduino:avr:nano:cpu=atmega328",
  "optimizeForDebug": false,
  "port": "COM6"
}
--------------------------
Warning: tool 'arduino:arduinoOTA@1.3.0' is not installed. It might not be available for your OS.Warning: tool 'arduino:avrdude@6.3.0-arduino17' is not installed. It might not be available for your OS.Upload error: Error: 2 UNKNOWN: uploading error: cannot execute upload tool: exec: "{runtime.tools.avrdude.path}/bin/avrdude": file does not exist

[image]

LumitoLuma, 2020-09-23:
Hello, @czghzc! Follow this steps (it worked for me for a very similar problem):

1.  Go to boards manager.
2.  Install *AVR Boards by Arduino* (version 1.82).
3.  Then, uninstall *AVR Boards by Arduino*.
4.  Finally, install the latest version of *AVR Boards by Arduino* (version 1.83).

Now try to upload the sketch.

Hope it works!

czghzc, 2020-09-24:
> 你好，@ czghzc！请执行以下步骤（它为我解决了一个非常类似的问题）：
> 
> 1. 去董事会经理。
> 2. 安装_Arduino_（版本1.82）的_AVR开发板_。
> 3. 然后，_通过Arduino_卸载_AVR开发板_。
> 4. 最后，安装_Arduino_的最新版本的_AVR开发板_（版本1.83）。
> 
> 现在尝试上传草图。
> 
> 希望它能起作用！

> Hello, @czghzc! Follow this steps (it worked for me for a very similar problem):
>
> 1. Go to boards manager.
> 2. Install _AVR Boards by Arduino_ (version 1.82).
> 3. Then, uninstall _AVR Boards by [...]

I followed your method: install 1.8.2-->uninstall-->install 1.8.3 without success.
Then I tried to install 1.6.23 and then installed 1.8.3 and succeeded.
thank you very much.

closed 2021-02-17 by ubidefeo